bambulab/BambuStudio #7719: Disable "Show online staff-picked models on the home page" doesn't persist between app launches.

deanocko, 2025-07-31:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

I work in a 3D printing lab at a university and we are soon to migrate to Bambu Lab printers for student printing. We have a strict, no printing of anything that resembles a weapon policy.

As such, I have tried to disabled the 'staff picked online models' that show up on the homepage as it is a distraction and often has models of guns on there as shown in the screenshots.

Despite disabling the setting in preferences as you can also see, whenever I re-open Bambu Studio the online models appear again on the homepage despite the setting being disabled. Initially when I disable the setting the models disappear but whenever I relaunch they appear again.

I think this is a bug. Please fix!

[image]
[image]

### Printer

NA

### How to reproduce

disable the 'staff picked online models' in preferences.
Close Bambu Studio
re-open, models will show again.

### Actual results

I work in a 3D printing lab at a university and we are soon to migrate to Bambu Lab printers for student printing. We have a strict, no printing of anything that resembles a weapon policy.

As such, I have tried to disabled the 'staff picked online models' that show up on the homepage as it is a distraction and often has models of guns on there as shown in the screenshots.

Despite disabling the setting in preferences as you can also see, whenever I re-open Bambu Studio the online models appear again on the homepage despite the setting being disabled. Initially when I disable the setting the models disappear but whenever I relaunch they appear again.

I think this is a bug. Please fix!

[image]
[image]

### Expected results

Disable setting and models will not be shown on homepage in perpetuity.

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

pgifford, 2025-08-01:
This is a dupe of
* #6058
* #6119
* #6376 
* #6484 
* #6865 

Opening a duplicate issue doesn't speed up development, better to add yourself to an existing issue.

MackBambu, 2025-08-05:
Fix in the next version.

JohnSmithCYM, 2025-08-14:
We have release https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.85 to solve this quesiton, please give it a try.

closed 2025-08-14 by JohnSmithCYM