bambulab/BambuStudio #7721: How to fix bambu studio very lag gui ?

Witawat, 2025-08-01:
### Bambu Studio Version

2.1.1

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 64bit

### Additional system information

Windows 11 64bit 
cpu i913980HX 
ram 32GB 
RTX4060 
ssd hdd 1 TB

### Printer

X1C

### How to reproduce

When I open the program, it hangs and stutters for a long time.

### Actual results

Various operations will hang for several seconds.
This includes selecting file loading options.

### Expected results

Don't want the GUI to lag or freeze like OrcaSlicer.

### Project file & Debug log uploads

All files will be the same.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2025-08-01:
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

Please follow the wiki and create DMP when ui is holding. Send DMPs to us by ticket or dropbox.

Witawat, 2025-08-01:
> https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue
> 
> Please follow the wiki and create DMP when ui is holding. Send DMPs to us by ticket or dropbox.
> […](#)

[log.zip](https://github.com/user-attachments/files/21545849/log.zip)