bambulab/BambuStudio #7724: Underextrusion after overhang

SinanAkkoyun, 2025-08-01:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

A1

### How to reproduce

Print part with overhang, outer/inner wall order at 0.06mm fine preset with 0.2mm

Observe underextrusion due to sudden speed change

[image]

### Actual results

[image]

### Expected results

No underextrusion

Perhaps do something like extrusion rate smoothing or something?

### Project file & Debug log uploads

(it's not a software related bug)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SinanAkkoyun, 2025-08-01:
For anyone encountering that issue: I am trying an Advanced -> Smooth coefficient of 0.5

qusheng-bambu, 2025-09-25:
> Hi, sorry for the late reply, I will test the new release and prepare a 3mf soon嗨，抱歉回复晚了，我会测试新版本并尽快准备一个 3mf

Hello, may I ask if you have reproduced this issue? If it still exists, please upload the 3mf file. We will investigate it as soon as possible. Thank you very much for your help