bambulab/BambuStudio #7729: Bambu Obxidian hot end will not allow ASA as filament preset

mfrenchtx-cloud, 2025-08-01:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

mac os sequoia 15.5

### Printer

Bambu X1C

### How to reproduce

I have installed the Bambu Obxidian complete hot end and updated the config, but when I load Bambu ASA in the AMS and sync it says filament not compatible with press sets, all presets are up to date but when I sync it tries to replace ASA filament with ABS filament settings even though the AMS recognizes the filament as ASA

[image]
[image]

### Actual results

after syncing AMS filaments which show ASA it changes to ABS

### Expected results

ASA filament should sync as ASA

### Project file & Debug log uploads

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Mujin-code, 2025-08-04:
Hello, thank you for your feedback. We have attempted to reproduce the issue based on the steps you described but were unable to do so. To help us investigate further,
1.Could you please provide the model project file along with a video showing the operation? You can use a simple screen recorder such as [ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1).
2.Could you pleaseprovide your log file so that we can further analyze the issue?
The steps to obtain the logs are as follows：https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log
This will help us check whether it’s a rendering issue or something else.
Thanks a lot for your help and patience!

mfrenchtx-cloud, 2025-08-09:
￼￼
On Aug 4, 2025, at 2:09 AM, Mujin_bambu ***@***.***> wrote:
￼attached is the requested information, the file is different than the original one but it happens on any file and I have tried several, so it is Bambu Studio related not stl/3mf file info related

> https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log