bambulab/BambuStudio #7731: Unable to log out of the account

MrSmile573, 2025-08-02:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

win11

### Printer

all

### How to reproduce

Unable to log out of the account; the logout button has disappeared from the homepage interface of BambooStudio.

### Actual results

Unable to log out of the account; the logout button has disappeared from the homepage interface of BambooStudio.

### Expected results

The account can be logged out, but the logout button on the Bamboo Studio homepage interface does not disappear.

### Project file & Debug log uploads

### Checklist of files to include

- [x] Log file
- [ ] Project file

Mujin-code, 2025-08-04:
Hi, thank you for reporting this issue!
To help us better understand and reproduce the problem on our side, could you please:
1.Check the version of bambu studio like this:

Mujin-code, 2026-01-08:
@Bamxcore hi , can you provide the log files and video for us to reproduce the problem , thank you ~

Mujin-code, 2026-01-08:
Got it, thank you. We will investigate the issue as soon as possible

Mujin-code, 2026-01-08:
@Bamxcore Hi, could you please provide us with the resolution and zoom size of your display screen

Bamxcore, 2026-01-08:
3440x1440

Zoom is at 100%