bambulab/BambuStudio #7734: the comma on the numpad is not working u always have to use the regular point

drc85, 2025-08-02:
### Bambu Studio Version

2.2.068

### Where is the application from?

Bambu Lab github releases

### OS version

windows 11 insider latest

### Additional system information

i7 8700k
32gb ram
rtx 4060

### Printer

H2D

### How to reproduce

just go in to resize, rotate or any other setting in the top bar...

it wont let you use the comma "," on the numpad!
it only accepts "." points!

### Actual results

it wont let you enter the "," comma 

### Expected results

we should be able to use the numpad "," too

### Project file & Debug log uploads

[logs.zip](https://github.com/user-attachments/files/21557844/logs.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

nblasgen, 2025-08-03:
Just 2 cents on this, but the issue is a localization issue.  Some countries use commas for decimals, some use points.  As Bambu Studio is a fork of SuperSlicer, which is a fork of PrusaSlicer, you might want to see if any of the forks before Bambu Studio also have the issue.  If they do, you can suggest the fix for them and then Bambu will automatically get that update.  While possible, I wouldn't expect Bambu to handle anything that isn't an issue with their fork (ie, anything custom they changed).  Again, just 2 cents here... feel free to ignore this comment.

drc85, 2025-08-10:
this can actually be, the offical version i have set to german, there the comma is working... on the beta i have it set to english to easy see wich is wich... but this sound logical..