bambulab/BambuStudio #7743: Smooth speed discontinuity bug

aldrin3210, 2025-08-03:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

P1s

### How to reproduce

I set the coefficient to .1 but the other side still have hard transition
[image]

The other side looks OK for oberhang transition

[image]

Speed settings

[image]

### Actual results

 only 1 corner is being smoothen.

### Expected results

Corver of overhang should have a smooth transition, i tried different values like 1-10 still the same but only 1 corner is being smoothen.

### Project file & Debug log uploads

[rangehood_hinge_proto_Prod_Fail.zip](https://github.com/user-attachments/files/21563898/rangehood_hinge_proto_Prod_Fail.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

aldrin3210, 2025-08-03:
Also when adding modifier will result in screen to be completely gray and unusable

Mujin-code, 2025-08-04:
Hello, thank you for your feedback. We have released the latest Beta version, which includes a fix for the login issue. You can download and install the updated software from the link below and check if the issue still persists. Thank you for your cooperation!
https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.68

aldrin3210, 2025-08-05:
it is now working. thanks!