bambulab/BambuStudio #7748: bambu studio app not working on my pc

brycestynm-gif, 2025-08-03:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home 24H2  5/‎5/‎2025  26100.4770

### Printer

Bambu lab A1

### How to reproduce

open app

### Actual results

nothing happens 

[image] #

### Expected results

the app should open and let me do stuff

### Project file & Debug log uploads

[New Compressed (zipped) Folder.zip](https://github.com/user-attachments/files/21568410/New.Compressed.zipped.Folder.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Mujin-code, 2025-08-12:
Hello, if you closed the software at that time, you could see the DMP file under the log file. If it was not generated, you can use this method to generate the DMP file the next time the software cannot be opened and send it to us for analysis.
Log query method：https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log
DMP：https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

brycestynm-gif, 2025-08-13:
Can you give more detailed instructions on how to find That
That answer doesn’t really help me with because I don’t have time to read
that I worked almost 7 full days a week working five full days a week

Mujin-code, 2025-08-13:
hi ，you can try this if you have time :
1.find the dmp file:
[image]

[image]
2.if you can't find the dmp file，you can try this way：
Open Windows Task Manager，and compress the dump file
[image]

thanks!