bambulab/BambuStudio #7765: Z shrinkage compensation

flyingbandit, 2025-08-05:
When using PAHT, I have discovered that the z axis shrinks by around 1%. Have not checked others yet.
For example, if you print a 119mm part, it is printed at 120.19mm tall on the Z axis. Upon completion and cooling, the part shrinks back to 119mm.

**Describe the solution you'd like**
A separate option to Shrinkage (only scales x-y), in the filament parameters for Z-axis compensation. This is because Z shrinkage is different to XY due to Z layers being non-uniform heat while cooling. This could be incorperated into a calibration per filament, ie you print a plate with an x-y and z at a suitable size (100mm perhaps) and then the slicer auto assigns this to the filament in the AMS slot.

**Describe alternatives you've considered**
I have to remember to scale parts by 101% on Z when using PAHT.

I have not checked other filaments yet.

flyingbandit, 2025-08-05:
[image]

Bambu studio on the left, Orca on the right

polygonfuture, 2025-08-08:
Yes, please add this!