bambulab/BambuStudio #7770: Slicing bug

neilcsinger, 2025-08-05:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 15.5

### Printer

Bambu Lab P1P

### How to reproduce

When I slice my model a tear appears in the print. The tear is not visible in the original model or the Prepare screen but the Slicer creates the tear. I repeated the steps in Orca Slicer v 2.3.0 and the tear never appears.

### Actual results

A tear appears in the model

### Expected results

a solid print

### Project file & Debug log uploads

[image]
[image]
[image]
[image]

[Archive.zip](https://github.com/user-attachments/files/21607853/Archive.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

neilcsinger, 2025-08-06:
The first image is from Bambu Studio in "Prepare". The last image is Bambu Studio after slicing in "Preview". The second and third images above are the same for Orca Slicer

Mujin-code, 2025-08-06:
Sorry, I can't see the picture you sent. It seems like the upload failed.

neilcsinger, 2025-08-06:
[image]

Mujin-code, 2025-08-06:
Hello, Hello, I tried slicing and printing with our latest version and everything went smoothly. We have released the latest Beta version. You can download and install the updated software from the link below and check if the issue still persists. Thank you for your cooperation!
https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.68

[image]

neilcsinger, 2025-08-20:
> Mujin-code

I tried the new release 2.2.0.85 and the bug in the slicing is still there. 

I then went back to the beta and it also has the bug.  Can you test it again?

Regards

  CONFIDENTIAL

 

--------------------------------
 
Neil Singer, PhD
President & CTO

AC Kinetics, Inc.
One Quarter Mile Road
Armonk, NY 10504
www.ackinetics.com
 
(914) 816-1221
Fax: (914) 273-2882