bambulab/BambuStudio #7771: Multidevice Printing Issue

CepheusSolutions, 2025-08-05:
### Bambu Studio Version

Beta Version 2.2.0.60

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

CPU: Intel Xeon CPU E3-1545M v5 @2.9GHz
48GB RAM
Intel Iris Pro Graphics P580

### Printer

Bambu X1 Carbon

### How to reproduce

Print to multiple devices 

### Actual results

The job pauses because it cannot receive the AMS mapping.

### Expected results

The Print job would print normally.

[log_iotc.txt](https://github.com/user-attachments/files/21608496/log_iotc.txt)

### Project file & Debug log uploads

S

[image]

[log_iotc.zip](https://github.com/user-attachments/files/21608503/log_iotc.zip)

[Shoe+Rack+Set+2.zip](https://github.com/user-attachments/files/21608509/Shoe%2BRack%2BSet%2B2.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

CepheusSolutions, 2025-08-05:
This issue has been occurring in the last few public releases of Bambu Studio.

XinZhangBBL, 2025-08-06:
Thank you for your feedback. We are looking in the issue. To get more information, could you reproduce the problem to offer log and video/screenshots for us. you can refer to https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log