bambulab/BambuStudio #7781: BambuStudio blocked by windows due to Behavior:Win32/CommandAndControl.B!np threat

mayaforce, 2025-08-07:
### Bambu Studio Version

2.2.0.68

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Windows 11
Real-time protection enabled under settings -> windows security -> virus & threat protection. 

### Printer

H2D, X1C, A1

### How to reproduce

Open BS
Connect to device
Start a print
Click on video feed for device (may not occur on first try. May need to click to a second device.)

### Actual results

Windows kills the BS program immediately. 

### Expected results

No death. 

### Project file & Debug log uploads

### Threat output:
Detected: Behavior:Win32/CommandAndControl.B!np
Status: Removed
A threat or app was removed from this device.
Date: 8/7/2025 2:57 PM
Details: This program is dangerous and executes commands from an attacker.
Affected items:
behavior: process: C:\Program Files\Bambu Studio\bambu-studio.exe, pid:11796:368007294745955
process:

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Mujin-code, 2025-08-08:
Hello, thank you for your feedback. We have attempted to reproduce the issue based on the steps you described but were unable to do so. To help us investigate further,
1.Could you please provide the model project file along with a video showing the operation? You can use a simple screen recorder such as [ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1).
2.Could you pleaseprovide your log file so that we can further analyze the issue?
The steps to obtain the logs are as follows：https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log
This will help us check whether it’s a rendering issue or something else.
Thanks a lot for your help and patience!

joekellyco, 2025-08-08:
I am seeing the same ,  Seems Bambu Studio is talking to known C2C IPs (IOCs/Command and Control for malware)  and is being blocked by Microsoft Threat Intelligence feed. 

[image]

spoonachario, 2025-08-08:
We're seeing similar via MDE - Virustotal/Palo/etc also showing the IP as malicious/RAT 

Our firewall is detecting the application flow to these IP IOCs as 'wyze'

Lot's of similar forums posts around wyze related IP's being marked malicious in the past.

Does BambuStudio interact with wyze cameras or their software somehow?

ilyasst, 2025-08-09:
Just got the same issue with Windows Defender.

pieterdevriesch, 2025-08-11:
Been having the same on my work network for the past week, mostly to ip 192.99.5.82 but also some other ones. It's clearly related to the p2p video stream. Since this IP has been on a list since 2022 already I suspect something was added on the Microsoft side. Still, not something you want to see in official software.

wmpa222, 2025-08-11:
Same here. 2 IPs every time I go to view a livestream

johntroupmagnixaero, 2025-08-11:
Seeing the same as well. It attempts a variety of ports on ip 66.206.1.250. The ip resolves to hivelocity servers. I don't see any relationship between them and Bambu. Bambu lists their infrastructure as AWS.