bambulab/BambuStudio #7782: Cannot Add SVG to Surface of Object

kimcharise, 2025-08-07:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.6 and Windows 11

### Additional system information

Windows: CPU: AMD Ryzen 3 2.4 Ghz
                  RAM: 8.0 GB
                  Display: 1920 X 1080 60 hz

Mac: CPU: M4
          RAM: 32 GB
          Display:  4480 x 2250 Retina

### Printer

Bambu X1C and Bambu H2D

### How to reproduce

1. Go to "Add SVG"
2. Check "Use Surface"
3. It crashes in macOS and hangs up in Windows

### Actual results

Full Report
-----------

{"app_name":"BambuStudio","timestamp":"2025-08-07 18:35:01.00 -0400","app_version":"02.01.01.52","slice_uuid":"b60d0c97-0a4a-3044-a1bb-ca6869105e32","build_version":"","platform":1,"bundleID":"com.bambulab.bambu-studio","share_with_app_devs":0,"is_first_party":0,"bug_type":"309","os_version":"macOS 15.6 (24G84)","roots_installed":0,"name":"BambuStudio","incident_id":"985FD1AE-529E-4FB1-9CCC-D52BD953E04B"}
{
  "uptime" : 37000,
  "procRole" : "Foreground",
  "version" : 2,
  "userID" : 501,
  "deployVersion" : 210,
  "modelCode" : "Mac16,3",
  "coalitionID" : 1335,
  "osVersion" : {
    "train" : "macOS 15.6",
    "build" : "24G84",
    "releaseType" : "User"
  },
  "captureTime" : "2025-08-07 18:35:01.5453 -0400",
  "codeSigningMonitor" : 2,
  "incident" : "985FD1AE-529E-4FB1-9CCC-D52BD953E04B",
  "pid" : 12258,
  "translated" : false,
  "cpuType" : "ARM-64",
  "roots_installed" : 0,
  "bug_type" : "309",
  "procLaunch" : "2025-08-07 18:30:36.9436 -0400",
  "procStartAbsTime" : 900021892009,
  "procExitAbsTime" : 906372010500,
  "procName" : "BambuStudio",
  "procPath" : "\/Applications\/BambuStudio.app\/Contents\/MacOS\/BambuStudio",
  "bundleInfo" : {"CFBundleShortVersionString":"02.01.01.52","CFBundleVersion":"","CFBundleIdentifier":"com.bambulab.bambu-studio"},
  "storeInfo" : {"deviceIdentifierForVendor":"BB079144-81E0-5970-AE85-EC9D62CEE826","thirdParty":true},
  "parentProc" : "BambuStudio",
  "parentPid" : 7151,
  "coalitionName" : "com.bambulab.bambu-studio",
  "crashReporterKey" : "B005B9FF-F8A0-37EB-6ADE-12030C0748CD",
[... 209 lines cut ...]
      "factorPackIds" : {

      },
      "deploymentId" : 240000004
    }
  ],
  "experiments" : [

  ]
},
  "reportNotes" : [
  "Extracting libpas PGM metadata failed."

### Expected results

The SVG should blend in with the body of the object

### Project file & Debug log uploads

[debug_Thu_Aug_07_18_30_37_12258.log.0.zip](https://github.com/user-attachments/files/21674152/debug_Thu_Aug_07_18_30_37_12258.log.0.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Mujin-code, 2025-08-11:
Hello, can you provide us with your 3MF and SVG files?

kimcharise, 2025-08-11:
Hello,

Please see attached files.
 Blank+NFL+Football+Helmet+with+stripes+-+separated2.3mf
<https://drive.google.com/file/d/1FXLHae_OBbyrLGALL-Depso2mR36K0Km/view?usp=drive_web>
I am trying to add the SVG to the sides of the helmet. I can attach the
SVG, but as soon as I try to use "curve to the body" that's when it hangs
up.

Mujin-code, 2025-08-11:
Hello, I am unable to view the content of the cloud storage as the website shows a refusal to view it

kimcharise, 2025-08-11:
The file is too large to send via email. Is there a place that I can upload
it to you?

Mujin-code, 2025-08-12:
Hi, I can download 3MF now, but there is no specific SVG file inside. I tried putting an SVG file myself:
1.Can you provide an SVG file of yours?
2.If I use the projection function, there will be a slight lag, but it will eventually return to normal. I would like to ask if it will be restored on your end? Have you ever tried waiting for a while? If your end keeps getting stuck, could you provide us with the dmp file? The generation method is as follows:https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

adele-with-a-b, 2026-09-06:
Correcting my comment above from 2026-05-22: **#10847 was reverted on 2026-05-28** ([`ced8934c`](https://github.com/bambulab/BambuStudio/commit/ced8934c7908780a73fe5303d0d66e676b36ade7)), so that fix is not in `master` and never shipped. What I wrote — "the underlying root cause is now fixed in main" — was true when I posted it and wrong six days later. Apologies for leaving it standing this long; anyone who read the thread and stopped watching came away with the wrong picture.

The stated reason for the rollback, from the reviewer on that PR: *"Our internal tests currently find that there is still a 50% chance of crashing."* That finding fits the diagnosis rather than refuting it. `corefine`'s recursion depth on near-degenerate input is data-dependent and effectively unbounded, so moving the worker stack from 4 MB to 16 MB shifts the threshold without establishing one — it converts some crashes into successes and leaves the rest untouched. It was a mitigation presented as a fix, and the internal test was right to catch the difference.

One data point for anyone tracking the same crash: OrcaSlicer cherry-picked the identical patch the same day ([OrcaSlicer#13772](https://github.com/OrcaSlicer/OrcaSlicer/pull/13772)) and still carries `set_stack_size(16 * 1024 * 1024)` in `Thread.hpp` today, three and a half months on.

Current state on `master`: `Thread.hpp` is back to the 4 MB default, and `CutSurface.cpp` has not been modified since 2024. This crash is still live.

Haidiye00, 2026-09-09:
https://github.com/bambulab/BambuStudio/releases
Hi, please download the latest beta version to try it out. If there are still issues, we can discuss them again. Thanks a lot.

closed 2026-09-09 by Haidiye00

adele-with-a-b, 2026-09-10:
Update, and a correction to my own earlier comments on this thread: **this is fixed in `master` as of 2026-09-01**, and my previous root-cause analysis was wrong.

Bambu root-caused it in [`e654e5af`](https://github.com/bambulab/BambuStudio/commit/e654e5afc865cec6187f71e0bb2bee967d8c563b) — *"FIX: fix macOS SVG/text emboss use_surface crash (GMP arm64 x18)"*. It was never a stack overflow.

**The actual cause:** GMP 6.2.1's arm64 `mpn` assembly uses register `x18`, which on Darwin/Apple Silicon is the *reserved platform register*. Clobbering it corrupts memory while CGAL runs its exact-arithmetic predicates during the surface cut. The fix applies upstream GMP changeset `5f32dbc41afc` — the same patch Homebrew and MacPorts carry — in the deps build, moving that code to x17/x6/x14. Per the commit, the rebuilt arm64 libgmp then passes GMP's full `make check`; **36 of 50 tests were failing on Apple Silicon before it**.

That explains several things my stack-overflow theory never did:

- Why it was macOS/Apple Silicon only. A 4 MB worker stack was fine on Windows and Linux, which makes no sense if recursion depth were the problem.
- Why the same input crashed only some of the time. Memory corruption from a clobbered register, not a depth that happened to vary run to run.
- Why `__stack_chk_fail` appeared in the crash reports. The stack canary was tripped *by* the corruption — I read that backwards as evidence of exhaustion.
- Why the 16 MB stack bump in #10847 helped partially but left Bambu's measured ~50% residual crash rate. A larger stack changes memory layout, which changes whether the corrupted write lands somewhere fatal. It was masking the bug rather than fixing it, and **the revert was correct**.

**Which builds have it:** I checked the release tags directly — `v02.08.03.66` (2.8.3 Public Beta, 2026-09-08) contains the commit; `v02.08.02.61` (2.8.2 Public Release, 2026-08-21) does not. So if you're on the current stable release and hitting this, the 2.8.3 beta is where the fix is. If you build from source, note the fix is in a *dependency* patch — you need a clean deps rebuild, not just an incremental app build, or you'll keep linking the old libgmp.

**One important caveat: if you're on Windows or x86_64 Linux, this fix is probably not yours.** `x18` is an arm64 register and it is only *reserved* on Darwin, so this particular corruption is specific to macOS on Apple Silicon. The patch is applied unconditionally because it's harmless elsewhere, but a crash in the same "Use surface" code path on Windows or on x86_64 Linux is a different defect and is worth its own report with a backtrace or crash dump attached.

Apologies for the churn on this thread. My earlier analysis was confident and wrong, and I'd rather say so plainly than leave it standing as the last word.