bambulab/BambuStudio #7788: Mesh Boolean Tool non-functional

ankonaskiff17, 2025-08-08:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home 24H2 Build 26120.5733

### Printer

P1S

### How to reproduce

Unable to do Subtraction, Intersection functions at all.  All that happens is it merges the two models shapes and throws a cantilever fault. Intersection is completely non-functional. Difference can only be done using negative part feature. It acts as if tool is disabled.
Also, wiki instruction for Mesh Boolean is out of date in wiki

### Actual results

Nothing happens

### Expected results

Expect Union, Difference, Intersection to work as described in wiki.
With this tool not functioning forces user to use an external CAD tool such as Fusion to do the function then reimport result back into Studio.

### Project file & Debug log uploads

This occurs prior to prior to project creation.
There are reports of this issue going back to 2024

### Checklist of files to include

- [ ] Log file
- [ ] Project file

qusheng-bambu, 2025-08-11:
Hello, I didn't reproduce your question using a simple block model.

1：Perhaps you can try checking "delete input" to see the result more intuitively

2: If there are still issues, could you please provide the 3mf file where the problem occurred and the related operation video? This can help us locate and solve the problem. Thank you very much!

[image]

qusheng-bambu, 2025-08-12:
Thank you very much for your suggestion! We will modify the relevant content of the wiki to enable users to have a more intuitive understanding of the usage steps of the tool.