bambulab/BambuStudio #7789: Issue while using Fill option for colour painting

TheSoumilArora, 2025-08-08:
### Bambu Studio Version

2.2.0.68

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Tahoe 26.0

### Additional system information

Nothing else running on the system.
Tried multiple times, everytime, when Fill option is selected in colour painting for multi-colour prints, and cursor moves over any face. (tried both edge detection and connected same colour.

Update- I did try another model and it works fine, but doesn't work for this model

### Printer

P1S

### How to reproduce

Load a model
Go to colour painting
Choose fill option
Hover cursor over any face

### Actual results

The app just crashes and closes

### Expected results

It should let me colour the face with the selected colour

### Project file & Debug log uploads

[WOPI keychain_v2.3mf.zip](https://github.com/user-attachments/files/21690534/WOPI.keychain_v2.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Haidiye00, 2025-08-12:
Hello, a new version is expected to be released in the next few days~

Haidiye00, 2025-08-13:
https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.85
Please use the latest version, thank you

closed 2025-08-13 by Haidiye00

TheSoumilArora, 2025-08-13:
Yes I saw the new version, I’ll update

Thank you for the prompt response!