bambulab/BambuStudio #7796: Won't stay logged in

ethicole, 2025-08-09:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

A1

### How to reproduce

1. verify username is displayed in upper left of screen
2. Select any online model
3. click download and open
4. Get prompted to log in (already logged in)
5. click google symbol or browser link
6. enter credentials at bambulab.com
7. Return to Bambu Studio

### Actual results

Endless loop of browser logins and still being prompted to log in. unable to download any models.

### Expected results

Log in successfully, download models

### Project file & Debug log uploads

### Checklist of files to include

- [ ] Log file
- [ ] Project file

smyserteejay, 2025-08-10:
having the same issue

qusheng-bambu, 2025-08-13:
Thank you very much for your feedback! I didn't reproduce this issue locally. This should be a probabilistic bug. Can it be solved by restarting Studio? Could you please upload the Studio log? This will help us troubleshoot this issue. We will do our best to solve it.

The steps to obtain the logs are as follows：
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log