bambulab/BambuStudio #7797: Add option to import MakerWorld models without filament/settings

father5boys, 2025-08-09:
Feature Request: Option to Import MakerWorld Files Without Overwriting Filament/Printer/Slicer Profiles
Summary:

When using the “Download and Open” button on MakerWorld, the downloaded .3mf file automatically launches in Bambu Studio with all of the original creator’s printer, filament, and slicer profiles applied. This behavior overwrites or clutters the user’s local preset libraries — even when those presets are irrelevant or incompatible with the user’s actual setup.

Problem:

There is no user-facing option to import geometry only from MakerWorld’s “Download and Open” workflow.

Disabling “import filament profiles” in Preferences is not available in Bambu Studio 2.1.x.

The “Clear my choice” option in Preferences does not restore the import dialog for MakerWorld-triggered files.

Some users have no interest in importing foreign presets but do want to access the geometry quickly from MakerWorld.

Impact:

Filament lists become cluttered with profiles the user does not own.

Printer and slicer settings are unintentionally overridden.

Workflow efficiency is reduced, as users must manually clean up after every import or avoid MakerWorld entirely.

Proposed Solution:

Add an Import Options Prompt for all .3mf project files opened via MakerWorld, regardless of prior “remember my choice” settings.

Include an “Import Geometry Only” checkbox (with “remember my choice” option).

Alternatively, allow MakerWorld to serve both the original .3mf and a plain STL download side-by-side.

Benefits:

Preserves the convenience of “Download and Open” while respecting user preferences.

Prevents unwanted profile imports and keeps local libraries clean.

Increases flexibility for users with custom or non-standard printing setups.

Example Workflow:

User clicks “Download and Open” in MakerWorld.

Bambu Studio prompts:

Import all settings and geometry

Import geometry only

User selects preferred behavior and can choose to remember it.

Feche, 2026-11-14:
When will this be implemented? it is a pain for advanced users, I have my profiles tuned and working - I do not need creators profile for printing. 
Please add the option under settings!

Feche, 2026-11-14:
@BambulabRobot and it is also related to https://github.com/bambulab/BambuStudio/issues/11359, so when are you going to fix/implement this? I need to use the printer the way I WANT TO, not like you want me to use it.