bambulab/BambuStudio #7804: Bambu Studio Crashing when Slicing Plate Every Time

jensstam, 2025-08-10:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: 3th Gen Intel(R) Core(TM) i7-13700K (3.40 GHz)
Memory: 32.0 GB
GPU: NVIDIA 4090
1TB nvm

### Printer

BambuLab A1

### How to reproduce

Using model Container Family by KL Worx, size D 70mm x H 50 mm
Using BambuLab Orange PLA Basic
Using default profile settings except change sparse infill pattern to Gyroid

Bambu Studio has also been crashing a lot when slicing other models but this one fails for me every time with these settings

### Actual results

Application crash during slicing (at around 80% mark)

### Expected results

Slice the model and let me print

### Project file & Debug log uploads

[container and lid 70x50(3).zip](https://github.com/user-attachments/files/21703463/container.and.lid.70x50.3.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Diatom-Banbu, 2025-08-19:
@jensstam Hello, we've carefully examined the log files and memory dump file you provided and have located the crash during the seam generation process. The crash occurred during matrix calculations during automatic seam generation, but no logic errors were found. Further investigation revealed that the crash occurred at different loop locations each time, more like a hardware failure. 
As we known, Intel's 13th and 14th generation CPUs with “K” have reportedly been defective and may be unstable. Therefore, we recommend checking your PC's stability. Due to the random nature of this hardware failure, we had difficulty reproducing the crash. Therefore, we would be very grateful if you could test whether your CPU is stable and let us know the results so that we can locate the cause of the crash more precisely.
Intel provides a processor stability checker that can verify the stability of your processor:  https://www.intel.com/content/www/us/en/download/15951/intel-processor-diagnostic-tool.html

Alternatively, as a workaround, you can try selecting random or back-up seam position. However, this is only a temporary fix and will not repair your hardware; your CPU may still gradually fail. 
[image]

jensstam, 2025-08-26:
According to the stability checker all is good in the CPU neighborhood.... also not having any other issues of CPU stability on the system other than with Bambu Studio

[image]

[TESTRESULTS.TXT](https://github.com/user-attachments/files/21978464/TESTRESULTS.TXT)

Diatom-Banbu, 2025-08-26:
@jensstam The test results do not indicate any known issues with the processor. However, given that hardware issues can be subtle and difficult to identify, they may only be triggered by certain instructions or algorithms, making verification difficult. We were unable to reproduce the crash you described on any of our test computers. Relying solely on the logs and memory dump file you provided, we've located the crash in automatic seam positioning. 
Please try using "random" seam positioning to see if it successfully slices. If only automatic seam positioning causes the crash, the algorithm library may be using certain instruction sets or CPU features, causing instability. We will continue to investigate.

jensstam, 2025-08-27:
Hi, I have updated to the latest version of Bambu Studio and it is no longer crashing when attempting to splice that model with the exact same specifications I provided. The seam position was left at Aligned and not changed to Random, but it still works after the update to the latest version.

Diatom-Banbu, 2025-08-29:
Thank you for your feedback. If you have any other questions or more information, please feel free to discuss with us and we will continue to pay attention.

closed 2025-09-01 by zhimin-zeng-bambulab