bambulab/BambuStudio #7822: When loading a 3D model, the model is not displayed, only a white canvas.

Dmitry-Chernikov, 2025-08-12:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Built from source by yourself or others

### OS version

Kernel: 5.15.0-151-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 5.4.12     tk: GTK 3.24.33 wm: Mutter dm: LightDM Distro: Linux Mint 21 Vanessa base: Ubuntu 22.04 jammy

### Additional system information

Компьютер
******************

Общая информация
-------------------------------

-Компьютер-
Процессор		: AMD Ryzen 9 5950X 16-Core Processor
Память		: 131804 Мб (3440 Мб занято)
Тип машины		: Настольный компьютер
Операционная система		: Linux Mint 21
Пользователь		: dmitry (dmitry)
Дата/время		: Вт 12 авг 2025 14:29:10
-Дисплей-
Разрешение		: 3840x1080 пикселей
Рендер OpenGL		: GeForce GTX 580/PCIe/SSE2
Поставщик X11		: The X.Org Foundation
-Аудио устройства-
Звуковая карта		: HDA-Intel - HDA NVidia
Звуковая карта		: HDA-Intel - HD-Audio Generic
-Устройства ввода-
 Power Button
 Power Button
 Microsoft Wired Keyboard 600
 Microsoft Wired Keyboard 600 Consumer Control
 Microsoft Wired Keyboard 600 System Control
 Logitech Wireless Receiver Mouse
 Asus WMI hotkeys
 Eee PC WMI hotkeys
 HDA NVidia HDMI/DP,pcm:3
 HDA NVidia HDMI/DP,pcm:7
 HD-Audio Generic Front Mic
 HD-Audio Generic Rear Mic
 HDA NVidia HDMI/DP,pcm:8
 HDA NVidia HDMI/DP,pcm:9
 HD-Audio Generic Line
 HD-Audio Generic Line Out Front
 HD-Audio Generic Line Out Surround
 HD-Audio Generic Line Out CLFE
 HD-Audio Generic Front Headphone
-Принтеры (CUPS)-
Canon-LBP710C-PPD		: <i>Default</i>
-SCSI диски-
ATA INTEL SSDSC2KW01

Операционная система
---------------------------------------

-Версия-
Ядро		: Linux 5.15.0-151-generic (x86_64)
Версия		: #161-Ubuntu SMP Tue Jul 22 14:25:40 UTC 2025
Библиотека C		: GNU C Library / (Ubuntu GLIBC 2.35-0ubuntu3.10) 2.35
Дистрибутив		: Linux Mint 21
-Текущая сессия-
Имя компьютера		: dmitry-System-Nutshell
Пользователь		: dmitry (dmitry)
[... 1413 lines cut ...]
### Printer

Bambu Lab X1 Carbon

### How to reproduce

1 Opening a 3D model

### Actual results

[BmbuLog.txt](https://github.com/user-attachments/files/21734443/BmbuLog.txt)

### Expected results

[image]

### Project file & Debug log uploads

[Open Bambu CMD LOG.txt](https://github.com/user-attachments/files/21734543/Open.Bambu.CMD.LOG.txt)
[BmbuLog.txt](https://github.com/user-attachments/files/21734486/BmbuLog.txt)

### Checklist of files to include

- [x] Log file
- [x] Project file

wolfiediscord, 2025-08-13:
Having the same issue. Seems to be due to the latest update. Reverting back to this commit seems to fix it for now.
```
89ea13aa4a965cfd6a5538503f905b16edc24a416e94d88f68cef734e7504807
```
Using Fedora Kinote 42 with KDE Plasma on Wayland. Haven't tested on X.Org, but according to others on the Discord server, it seems to work under X.Org.

MackBambu, 2025-09-05:
> This is rather unfortunate. Any ETA on when Wayland support will be back? 
@guanyun-gudujian

Dmitry-Chernikov, 2025-09-06:
> Это довольно неудачно. Кто-нибудь из ETA о том, когда вернется поддержка Wayland? 5 сентября 2025 года, в 07:45, MackBambu ***@***.***> написал: МакБамбу оставил комментарий [(бамбулаб/БамбуСтудио#7822)](https://github.com/bambulab/BambuStudio/issues/7822)
> Из-за проблем, связанных с водителем, поддержка была возвращена из Wayland на X11, что может повлиять на некоторые из предыдущих настроек.
> [88э91д](https://github.com/bambulab/BambuStudio/commit/88ae91da7d8f6b619d71ad7af1963438cdd6d85c)
> 
> —Ответите на это письмо напрямую, просмотрите его на GitHub или unsubscribe.Вы получаете это, потому что вы прокомментировали.Опозор ID: ***@***. ***>

I wanted to check on Wayland, but unfortunately I couldn't do it, I managed to install Linux Mint version 21, since there is no driver for the Nvidia-390-driver video card in the repository. New versions of Linux Mint do not start the X server at all. I tried in compostable mode, the interface slows down a lot, freezes and reboots.