bambulab/BambuStudio #7823: Option to Prioritize Custom Printer and Filament Profiles Over Defaults in Bambu Lab Studio

L0rdS474n, 2025-08-12:
Having custom printers or filament profiles in Bambu Lab Studio currently does not integrate smoothly with profile selection. If I have a customized printer profile (e.g., H2D with heat soak enabled) and load a project that requests an H2D, Studio will choose the default H2D profile instead of my customized one. The same issue occurs with filaments: if I have a customized profile for a specific filament — for example, blue TPU 95A HF — the synchronization process still selects the default TPU 95A HF profile rather than my customized version.

This creates unnecessary extra steps, especially when working with multiple printers or large AMS setups. For example, with 4 AMS2 Pro units fully loaded with different filaments, the system could match the requested filament and, if a color is specified, automatically select the correct customized profile. Instead, I currently have to manually switch to my custom profile every time the printer or filament changes.

The proposed solution is to add an option in the Studio settings to control this behavior: a checkbox allowing the user to choose whether Studio should default to using custom printer/filament profiles (when available) or always use the default profiles. This would make workflows far more efficient for advanced setups, while still allowing users who prefer defaults to keep that behavior.

Alternatives considered: manually changing to the correct custom profile every time, which is inefficient and prone to mistakes, especially in multi-printer or multi-filament workflows.

Additional context: this change would significantly improve the user experience for anyone with customized printers or filaments, particularly in professional or high-throughput environments where print settings are carefully tuned and need to be preserved automatically.

hotellonely, 2025-08-12:
This is a must have, I live in apartment that is quite not really sound proof. Everytime I forget to turn off mech check using my customised profile, I'm worried that my neighbour would eventually remove me like how I removed my other printers.

I want to be able to:
1. disable nozzle clumping detection by default
2. disable mech check / input shaper check before each print

**(I'm also okay if Bambu can move the options to before print send task window, actually that would be even better)**

cforsyth68, 2026-05-24:
Yes, Yes, Yes, and YES! We need this, it is so frustrating to have to manually set defaults each time I switch between my multiple printer types.