bambulab/BambuStudio #7834: Small text loses formatting after update to 2.2.0.85

cameroncondry, 2025-08-13:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 14.7.4 (23H420)

### Printer

Bambu Lab A1 Mini

### How to reproduce

1. Go to the Object view
2. Go to Plate 1 > Swatch (Text) > type
3. Update the text shape
4. Change any text value

### Actual results

The text loses formatting and centering, and the space character becomes much larger. It also doesn't go back to the previous formatting once it's lost. 

### Expected results

The text should be updated without losing the formatting.

### Project file & Debug log uploads

[filament-swatches.3mf.zip](https://github.com/user-attachments/files/21746694/filament-swatches.3mf.zip)

This is my test file that I print after calibrating a new filament. Previously, I could update the object text and print the new filament without issue. However, if any edits are made to the object text, the formatting is completely lost.

### Checklist of files to include

- [ ] Log file
- [x] Project file

cameroncondry, 2025-08-13:
Thank you for the response! I was able to get the text very close to its original form using the following:

Thickness: 3.9
Text Gap: -1.10

For the first line, I need an extra space before the text, and the second line needs an extra space after the text to center the text correctly. I can also add an X offset, but I don't know how to precisely calculate the space character with extra space reliably.

Please let me know if there are any other helpful details that I can provide. Thanks!

cameroncondry, 2025-08-13:
Another quick case... In this case, I can't get the first and second line to center with each other without adding in a custom negative x offset. Additionally, the space adds a lot of negative space in both lines, even though I'm in a negative text gap.

[image]

Haidiye00, 2025-08-14:
Well, I suggest you use the previous official version so we can evaluate how to solve it first. Currently, many new and old text tools are unable to achieve compatibility in certain areas.

closed 2025-08-27 by Haidiye00