bambulab/BambuStudio #7835: Text Tool Bug

maskinazi, 2025-08-13:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

Macos 15.6

### Additional system information

Mac mini M4 Pro
64GB Memory

### Printer

Bambu Lab P1S

### How to reproduce

1. Select Plate 17 (Labels)
2. Select the Label M3-16
3. Clone this items
4. Select the Text in the cloned item
5. Click on the Text Tool
6. Change the Input Text to M3-20

### Actual results

The text in the changed item  is not spaced the same as the original item (see screenshot)

[image]

 is not spaced the same as the original item (see screenshot)

### Expected results

I expect the format of the text in the edited text to be formatted with the same spacing the original text

### Project file & Debug log uploads

[Under Desk Drawers .3mf.zip](https://github.com/user-attachments/files/21746761/Under.Desk.Drawers.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Haidiye00, 2025-08-16:
https://github.com/Haidiye00/BambuStudio/actions/runs/16987363879

[image]
Please try this,thanks

Haidiye00, 2025-08-27:
Hello, please upgrade to the latest version
https://github.com/bambulab/BambuStudio/releases 

If there are still problems, we can solve them

GuyH77, 2025-08-27:
> I too have noticed this issue. I also note the text handle position may also be incorrect. Centring text now longer works nor can I use object position to place it correctly at 0,0 of the object in X and Y.

@Haidiye00 Thank you for the improvements with the latest version. There is still a problem with the object coordinates. If I use 'center' then the text is placed in the center of the object. However, I often use the object coordinates (great addition) and the handle for the text is not central, therefore object coordinates when set to 0,0 differs from when using the center command. This should not be the case. They should place the text in the same location.

Center

[image]

Object coordinates 0,0

[image]

Haidiye00, 2025-08-28:
Okay, let's optimize it later