bambulab/BambuStudio #7836: Blank page on prepare and preview tab since update to 2.2

bfrigon, 2025-08-13:
### Bambu Studio Version

2.2

### Where is the application from?

Bambu Lab github releases

### OS version

Linux Bazzite 42, Flatpak

### Additional system information

Bazzite 42
AMD RX9060 GPU

### Printer

X1C

### How to reproduce

It was working fine with the flatpak package but since the update to 2.2, it is no longer usable. Tried to delete config and reinstall, same issue.

I always get a blank page on the prepare and preview tab

[image]

### Actual results

From the command line, i get these errors (flatpak run com.bambulab.BambuStudio)

`[2025-08-13 03:21:16.691917] [0x00007f57e7cb3480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
Gtk-Message: 03:21:16.740: Failed to load module "colorreload-gtk-module"
after dlopen, network_module is 0x5583aed11ca0

(bambu-studio:2): Gtk-CRITICAL **: 03:21:17.179: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 03:21:17.179: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 03:21:17.179: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:2): Gtk-CRITICAL **: 03:21:17.180: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 03:21:17.180: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 03:21:17.180: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:2): Gtk-CRITICAL **: 03:21:17.181: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 03:21:17.181: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:2): Gtk-CRITICAL **: 03:21:17.181: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:2): Gtk-CRITICAL **: 03:21:17.182: gtk_widget_set_size_request: assertion 'width >= -1' failed
`

### Expected results

The normal content on prepare and preview tab

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/21749017/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

guanyun-gudujian, 2025-08-13:
@bfrigon 
pls see this comment and have a try
thanks
https://github.com/bambulab/BambuStudio/issues/7343#issuecomment-3086161951

MelvinSnijders, 2025-08-16:
Same issue here!

MelvinSnijders, 2025-08-30:
Any update on this?

qusheng-bambu, 2025-08-30:
Yes, you can refer to the issue below. The new flatpak has been resolved，If you have any further questions, please contact us
https://github.com/bambulab/BambuStudio/issues/7843
https://github.com/bambulab/BambuStudio/issues/7880

MelvinSnijders, 2025-08-30:
Amazing, thanks! Will check it out 🙌🏻