bambulab/BambuStudio #7842: Text tool space character creates too wide gaps after recent update

hpdrk08, 2025-08-13:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab github releases

### OS version

windows 111

### Additional system information

amd ryzen 5 5600.
msi rtx 4060 8g
32gb ram 3200mhz

### Printer

a1

### How to reproduce

1.Open the text tool.
2.Type any text with spaces, e.g. 61 HL 053.
3.Compare the gap between characters with a version that uses dashes instead, e.g. 61-HL-053.
4.Notice that spaces create much wider gaps than expected.

### Actual results

61 HL 053   ← spaces are too wide  
61-HL-053  ← normal spacing

[image]

### Expected results

Spaces between words should have normal, consistent width — the same proportion as before the update and matching standard font spacing — so that 61 HL 053 looks visually balanced like 61-HL-053.

### Project file & Debug log uploads

[platecust.zip](https://github.com/user-attachments/files/21754059/platecust.zip)

[log.zip](https://github.com/user-attachments/files/21754076/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

tgroven, 2025-08-13:
Yes, this is so annoying!

Haidiye00, 2025-08-14:
Well, I suggest you use the previous official version. Let's first evaluate how to solve it. Currently, many new and old text tools are unable to achieve compatibility in certain areas.

Haidiye00, 2025-08-15:
[image]
Hello, I have identified the issue and have conducted preliminary testing. I will provide a package later to help you try it out. Thank you

Haidiye00, 2025-08-15:
https://github.com/Haidiye00/BambuStudio/actions/runs/16987363879

[image]
Please try this,thanks

mtnears, 2025-08-16:
So is there a way for me to use this with the current production release?  Can I add just a specific module?  If I use the whole download version, it is not signed and it doesn't let me print.

penanot, 2025-08-17:
Thank you. Now the text is perfect. Thanks @Haidiye00

NathanTarbox, 2025-08-17:
@mtnears You can create the text in the unsigned version, save it, then open and print in the normal version.  Just don't try to edit the text again in the current normal version or you'll end up reverting those changes.

I also tried it.  Worked as desired after some tinkering.
The new text feature did have some issues cutting off parts of letters when using 'cut' on a concave surface and using Surround Surface or Surround + Horizontal, but Surround projection by char avoided this issue.  It's possible that the  surface I was using had an elliptical rather than constant/radius curve, but that's just a guess.
I tried to resolve by using an increased thickness and negative embedded depth (to allow it to punch the lettering out through the object wall) but it wouldn't allow a negative embedded depth.

"Embedded" is also written as "Embeded".

Haidiye00, 2025-08-21:
The centering issue of text will be fixed in the next version, and a new version is expected to be released this month

Haidiye00, 2025-08-27:
Hello, please upgrade to the latest version
https://github.com/bambulab/BambuStudio/releases 

If there are still problems, we can solve them

closed 2025-08-27 by Haidiye00

dgtalmobile, 2025-08-29:
Hi Haidiye,

updated to the latest, centering fonts fix yehey. Thank you.

Can you help me with this, if its possible 

Scenario 

I got P1s combo and my ams is updated thru p1, but when I transfer to my X1E is say it need to update but the fw is lower in sequence. And x1e fw is still old. Btw fyi when I bought this x1e the lcd logo picture is x1c and it won’t connect to cloud but this is fix when I upgraded the fw manually and the logo is x1e and is connected within seconds.

I have a little question in printing a coster (ironing top layer) when printing only 1 in the bed it print flawessly (left pic) , but when I fill the bed with copies 4pcs when it printed the top layer is not that smooth (right pic) this was both printed in X1e

￼

But for p1s c 4 pcs in bed plate same file printed a bit more refind but not like single printing ( right pic)
Left pic is from X1e

Both printers using same polymer polysonic pro filaments but there’s no project filaments listed and I’m just using poly terra settings
 @.16 high quality .04mm, aranche, top layer ironing, top surface monotonic, gyroid. Is there something wrong with the printer alignment or my settings? 

￼

And another, when I’m using non bambupalte with special design sometimes it printed the 1st bottom layer a bit corse some lines, even bisque carbon fiber plate. But I always check bed alignment, is there a setting I need to change?

Btw x1e won’t recognized the cool plate super tack , keeps getting error, can’t find 

Sincerely 
Philip