bambulab/BambuStudio #7843: Bambu studio not showing anything on preview or prepare screens

Enderred2, 2025-08-13:
### Bambu Studio Version

2.1.1 Public Release

### Where is the application from?

Bambu Lab github releases

### OS version

SteamOS 3.8 Build: 20250807.1000 KDE Plasma Version: 6.4.3

### Additional system information

Operating System: SteamOS 3.8 Build: 20250807.1000
KDE Plasma Version: 6.4.3
KDE Frameworks Version: 6.16.0
Qt Version: 6.9.1
Kernel Version: 6.15.8-valve1-2-neptune-615-g49248f4e2ad1 (64-bit)
Graphics Platform: X11
Partition: B
Processors: 8 × AMD Custom APU 0405
Memory: 16 GiB of RAM (14.5 GiB usable)
Graphics Processor: AMD Custom GPU 0405
Manufacturer: Valve
Product Name: Jupiter
System Version: 1

### Printer

bambu lab a1

### How to reproduce

1. Launch the applicaion
2. Click on "Create new project" or click on a existing file

### Actual results

Nothing is projected onto the prepare or preview tabs, no model is being show, no print bed nor the settings that allow you to modify the model on the built plate.

[image]

### Expected results

The intended result is suppose to show the entire left side of left side of the screen showing the built plate and the settings to modify the model.

[image] 

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/21754337/log.zip)

[Untitled.zip](https://github.com/user-attachments/files/21754339/Untitled.zip)

In attempt to fix this problem, all save data was deleted in the process.
(To clarify a potentially immediate concerns, the OS has never been a problem when it come to these types of activities nor has the hardware)

### Checklist of files to include

- [x] Log file
- [x] Project file

wldcld, 2025-08-13:
I have the same problem, it started today.
### Bambu Studio Version
2.2.0.85
### Where is the application from?
Flathub
### OS version
Fedora Linux 42 (Workstation Edition) x86_64
### Additional system information
Kernel: 6.15.9-201.fc42.x86_64 
DE: GNOME 48.4 
WM: Mutter 
WM Theme: Adwaita 
CPU: AMD Ryzen 9 7950X (32) @ 5.883GHz 
GPU: AMD ATI 12:00.0 Raphael 
GPU: AMD ATI Radeon RX 7900 XT/7900 XTX/7900 GRE/7900M
### Printer
Bambu Lab X1C
### How to reproduce
1. Launch the applicaion
2. Click on "Create new project" or click on a existing file
### Additional Information
Before Bambu Lab showed this problem I upgraded a couple of packages (there was not upgrade for Bambu Studio installed or available). There were some libraries and drivers for the graphic card upgraded. Maybe there is a compatibility issue with these new versions?
cpp-0:15.2.1-1.fc42.x86_64
gcc-0:15.2.1-1.fc42.x86_64
gcc-plugin-annobin-0:15.2.1-1.fc42.x86_64
gcc-c++-0:15.2.1-1.fc42.x86_64
libstdc++-devel-0:15.2.1-1.fc42.x86_64
libstdc++-0:15.2.1-1.fc42.x86_64
libstdc++-0:15.2.1-1.fc42.i686
fluidsynth-libs-0:2.4.7-1.fc42.x86_64
libatomic-0:15.2.1-1.fc42.x86_64
libatomic-0:15.2.1-1.fc42.i686
libgcc-0:15.2.1-1.fc42.x86_64
libgcc-0:15.2.1-1.fc42.i686
libgfortran-0:15.2.1-1.fc42.x86_64
libgomp-0:15.2.1-1.fc42.x86_64
libgomp-0:15.2.1-1.fc42.i686
libquadmath-0:15.2.1-1.fc42.x86_64
mesa-dri-drivers-0:25.1.7-1.fc42.x86_64
mesa-filesystem-0:25.1.7-1.fc42.x86_64
mesa-libgbm-0:25.1.7-1.fc42.x86_64
mesa-libGL-0:25.1.7-1.fc42.x86_64
mesa-libEGL-0:25.1.7-1.fc42.x86_64
mesa-dri-drivers-0:25.1.7-1.fc42.i686
mesa-libOpenCL-0:25.1.7-1.fc42.x86_64
mesa-libgbm-0:25.1.7-1.fc42.i686
mesa-libGL-0:25.1.7-1.fc42.i686
mesa-libEGL-0:25.1.7-1.fc42.i686
mesa-filesystem-0:25.1.7-1.fc42.i686
mesa-vulkan-drivers-0:25.1.7-1.fc42.x86_64
mesa-va-drivers-0:25.1.7-1.fc42.x86_64
mesa-vulkan-drivers-0:25.1.7-1.fc42.i686
mesa-va-drivers-0:25.1.7-1.fc42.i686
mesa-libxatracker-0:25.1.7-1.fc42.x86_64
nodejs-undici-0:7.13.0-1.fc42.noarch
openjpeg-0:2.5.3-8.fc42.x86_64
rest-0:0.10.2-1.fc42.x86_64
selinux-policy-0:42.5-1.fc42.noarch
selinux-policy-targeted-0:42.5-1.fc42.noarch
toolbox-0:0.2-1.fc42.x86_64

Meroin, 2025-08-13:
I also have the same problem when trying to print today.

**Bambu Studio Version**

2.2.0.85

Where is the application from?
Flathub

**OS version**
"Nobara Linux 42 (KDE Plasma Desktop Edition)" (64 bit)

**Additional system information**

Kernel: 6.15.8-200.nobara.fc42.x86_64
CPU: AMD Ryzen 7 7800X3D 8-Core Processor
GPU: AMD Radeon RX 6900 XT
3.63, 6.15.8-200.nobara.fc42.x86_64)
Driver Version: 4.6 (Compatibility Profile) Mesa 25.2.0

**Printer**

Bambu Lab P1s

**How to reproduce**

    Launch the applicaion
    Click on "Create new project" or click on a existing file

pmc040, 2025-08-13:
Same problem with Linux Mint 21.3 cinnamon - the plate has disappeared. The Windows 11 versions appears to be working OK.

cpetry, 2025-08-13:
Yes, I can also confirm on Linux Mint cinnamon. Since latest update

asheiduk, 2025-08-13:
Same problem here after Bambulab Studio has been updated.

Bambulab studio: 2.2.0.85 from Flathub

CPU: 8-core AMD Ryzen 7 7800X3D (-MT MCP-) speed/min/max: 735/545/5050 MHz
Kernel: 6.8.0-65-generic x86_64 Up: 1h 55m Mem: 3960.1/63912.4 MiB (6.2%)
Graphics:
  Device-1: AMD driver: amdgpu v: kernel
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: amdgpu,ati
    unloaded: fbdev,modesetting,radeon,vesa gpu: amdgpu
    resolution: 2560x1440~60Hz
  OpenGL: renderer: GFX1101 (gfx1101 LLVM 15.0.7 DRM 3.57 6.8.0-65-generic)
    v: 4.6 Mesa 23.2.1-1ubuntu3.1~22.04.3

fehlfokus, 2025-08-13:
Same Problem here, Fedora 42, BambuStudio installed from flathub.
Is there a workaround?

TomSievers, 2025-08-13:
I experienced the same issue on Fedora 42 when installed from flathub. By downgrading to 2.0.3 I can see the preview again. For anyone else wanting to downgrade to 2.0.3 use the command below.

`sudo flatpak update --commit=315e88029ecc5e8da687125b2998b8e31dbe79b9fccba8712f0be36d3353560e com.bambulab.BambuStudio`

System Info:
  Kernel: 6.15.9-201.fc42.x86_64 arch: x86_64 bits: 64
  Desktop: GNOME v: 48.4 Distro: Fedora Linux 42 (Workstation Edition)
CPU:
  Info: 8-core AMD Ryzen 7 PRO 8840U w/ Radeon 780M Graphics [MT MCP]
Graphics:
  Device-1: Advanced Micro Devices [AMD/ATI] HawkPoint1 driver: amdgpu v: kernel
  Display: wayland server: X.Org v: 24.1.8 with: Xwayland v: 24.1.8
    compositor: gnome-shell driver: dri: radeonsi gpu: amdgpu
    resolution: 1920x1200~60Hz
  API: OpenGL v: 4.6 vendor: amd mesa v: 25.1.7 renderer: AMD Radeon 780M
    Graphics (radeonsi phoenix LLVM 20.1.8 DRM 3.63 6.15.9-201.fc42.x86_64)
  Info: Tools: api: glxinfo x11: xdriinfo, xdpyinfo, xprop, xrandr

flamingm0e, 2025-08-13:
Thanks for the workaround and downgrade command.

guanyun-gudujian, 2025-08-14:
do not use any unoffical workround

pls see this comment:
https://github.com/bambulab/BambuStudio/issues/7343#issuecomment-3086161951
thanks

guanyun-gudujian, 2025-08-14:
> > do not use any unoffical workround
> > pls see this comment: [#7343 (comment)](https://github.com/bambulab/BambuStudio/issues/7343#issuecomment-3086161951) thanks
> 
> Do you provide .desktop files for this? Why is the .AppImage labeled "ubuntu"? Can you see the confusion this causes? Why is Flatpak not supported officially?

it's meaningless to discuss these question here.
make it work and enjoy the 3d printing is the final purpose

guanyun-gudujian, 2025-08-14:
> > > > do not use any unoffical workround
> > > > pls see this comment: [#7343 (comment)](https://github.com/bambulab/BambuStudio/issues/7343#issuecomment-3086161951) thanks
> > > 
> > > 
> > > Do you provide .desktop files for this? Why is the .AppImage labeled "ubuntu"? Can you see the confusion this causes? Why is Flatpak not supported officially?
> > 
> > 
> > it's meaningless to discuss these question here. make it work and enjoy the 3d printing is the final purpose
> 
> ??
> 
> I can't double click a file and open it because it's an AppImage that isn't registered with the system.
> 
> I can't "Open With" because it's an AppImage that isn't registered with the system.
> 
> There is no icon for it in apps because it's an AppImage with no .desktop file.
> 
> The Flatpak version is literally put out and linked by Bambulab on the page...that's the version that people are installing.
> 
> Your dismissal of these basic concerns and functions is a little ridiculous.

ok  my stupid.
your os info pls?
we will have a look

Enderred2, 2025-08-14:
> ??
>
> I can't double click a file and open it because it's an AppImage that isn't registered with the system.
>
> I can't "Open With" because it's an AppImage that isn't registered with the system. [...]

I encountered a similar issue where the appimage wont run but the odd part being that my device is already compatible with this file type. 
It also may be that I don't know how to fully use Linux

guanyun-gudujian, 2025-08-14:
> > > ok my stupid. your os info pls? we will have a look
> > 
> > 
> > OS: Fedora Linux 42 (Workstation Edition) x86_64 Host: 20N2A01UUS (ThinkPad T490) Kernel: Linux 6.15.9-201.fc42.x86_64 Uptime: 3 hours, 25 mins Packages: 2055 (rpm), 29 (flatpak) Shell: zsh 5.9 Display (LG HDR WFHD): 2560x1080 @ 60 Hz in 34" [External] Display (LG HDR WFHD): 2560x1080 @ 60 Hz in 34" [External] DE: GNOME 48.4 WM: Mutter (Wayland) WM Theme: Adwaita Theme: Adwaita [GTK2/3/4] Icons: Adwaita [GTK2/3/4] Font: Adwaita Sans (11pt) [GTK2/3/4] Cursor: Adwaita (24px) Terminal: kitty 0.42.2 Terminal Font: NotoSansMono-Regular (11pt) CPU: Intel(R) Core(TM) i7-8665U (8) @ 4.80 GHz GPU: Intel UHD Graphics 620 @ 1.15 GHz [Integrated] Memory: 8.03 GiB / 15.24 GiB (53%)
> 
> appimage should be work on fedora system
> 
> pls drag & drop the appimage into the terminal and press enter key
> 
> then past the log here thanks

@Enderred2   have a try

flamingm0e, 2025-08-14:
> appimage should be work on fedora system
>
> pls drag & drop the appimage into the terminal and press enter key
>
> then past the log here thanks

I never said the AppImage wouldn't run.  I think you are misunderstanding the issue here.  I see that SEVERAL others have opened similar issues within the last couple of days. That alone should tell you this problem is bigger than just telling someone to run a stupid AppImage.

Here are the problems with running the AppImage:
1. it ignores my desktop theme and forces light theme
2. it doesn't register in the system as being able to open stl or 3mf files
3. it doesn't provide a .desktop file for it to be available as an application through your normal app launcher/home screen
4. it literally says Ubuntu in the filename which to anyone that isn't running Ubuntu makes them think it's not for them

Now that we're clear on what the problems are with AppImage, let's talk about how Bambu Studio is distributed.
Every mainstream distro is using Flatpak in their respective "App Store", aside from Ubuntu, which defaults to SNAPS.  If a user is looking for Bambu Studio, they open up the Software Store on their computer, and there it is...Bambu Studio, served up as a Flatpak, literally from Bambulab developer (not a third party).  This version that is published on Flathub is broken.  PERIOD.  This version that is being published on Flathub either needs to be REMOVED with information on how to get the same experience with the AppImage, or it needs to be fixed to prevent more bug reports of the exact same thing.  I even found a Bambulab web page last night that directed users to the flathub version from the website.

If you want the AppImage to be the standard for Linux users:
1. remove the "ubuntu" tag in the name
2. provide information on making files open with the AppImage (.stl, .3mf, etc)
3. provide information on making a .desktop file for it to show up in user's application lists
4. let the AppImage use the user's desktop theme choice
5. get rid of the Flathub flatpak version

There is a huge problem and you need to solve it in an official manner.

asheiduk, 2025-08-14:
Adding to @flamingm0e:

Another advantage of the Flatpak version are the automatic updates -- done by the same framework as the rest of the system. This convenience and consistency is quite valuable!

The Flathub version is mentioned in several release announcements of this very project ([one example](https://github.com/bambulab/BambuStudio/releases/tag/V02.00.03.54)). So I concur that -- from the outside -- it has the blessing of Bambulab. The link mentions >127k downloads -- I think quite a good chunk of the Bambulab userbase and -- I think -- most of the Linux users of that chunk.

IlikePepsi, 2025-08-15:
Same issue here with the flatpak version of BambuStudio on bazzite (fedora-kinoite).

flamingm0e, 2025-08-17:
As of right now, the latest version in Flatpak is working for me now.