bambulab/BambuStudio #7850: cannot use 0.001 step size in flow dynamics calibration

ftc2, 2025-08-13:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 15.6

### Additional system information

there appears to be an error in your input checking for the `Value step` field in flow dynamics cal. it does not let you use `0.001` step size.

### Printer

P1S

### How to reproduce

- do flow dynamics cal
- put `0.001` for `Value step`
- observe that it won't let you do it despite the stated input range being `>= 0.001`

### Actual results

cal does not proceed
[image]

### Expected results

cal proceeds

### Project file & Debug log uploads

tell me if you need this

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ftc2, 2025-09-20:
well, i confirm it is fixed for the pattern method with [0, 0.05] 0.001 step. thank you.

i can't test the line method right now, but i agree that the error message shown in @frenchiveruti 's screenshot could be a lot more helpful.

closed 2025-09-20 by ftc2

qusheng-bambu, 2025-09-22:
> > We have fixed this issue in the latest version. You can verify it with the latest version. If the verification is successful, you can close this issue我们已在最新版本中修复了此问题。您可以使用最新版本进行验证。如果验证成功，您可以关闭此问题
> > https://github.com/bambulab/BambuStudio/releases/tag/v02.02.01.60
> 
> Latest version is even less descriptive.最新版本的描述性更少。
> 
> <img alt="Image" width="1032" height="714" src="https://private-user-images.githubusercontent.com/1095997/491861435-6cc4f6f4-ae67-45df-bdac-c7858850933e.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTg1MDk4NDUsIm5iZiI6MTc1ODUwOTU0NSwicGF0aCI6Ii8xMDk1OTk3LzQ5MTg2MTQzNS02Y2M0ZjZmNC1hZTY3LTQ1ZGYtYmRhYy1jNzg1ODg1MDkzM2UucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDkyMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA5MjJUMDI1MjI1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODg4OWY4Mjk0OWY1YjU3OGZiYWExZmRlZDZkYzA0MjRhNTA5OTU0Y2M5ZTA0NzE1ZTJkY2I0MTQ0ZTg5YzRhNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.IZRz_QNpcbm_cDWpZ1KKxYOucn8wrixfpHJhBeT5Ttk"> <img alt="Image" width="1267" height="865" src="https://private-user-images.githubusercontent.com/1095997/491861442-a3cfd746-713f-456d-84b1-ee2fa7184e02.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTg1MDk4NDUsIm5iZiI6MTc1ODUwOTU0NSwicGF0aCI6Ii8xMDk1OTk3LzQ5MTg2MTQ0Mi1hM2NmZDc0Ni03MTNmLTQ1NmQtODRiMS1lZTJmYTcxODRlMDIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDkyMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA5MjJUMDI1MjI1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzA5N2M1ZTcwYzA3Zjc0NTRlYTk2ZDA4YWU3NmEzMzFiNTMxNmFmOTBhMDQzMGFmODE4ODY5M2E0MWU0ZWUyMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.PaY4Va23kT3JkGbSHKEFb_IZRLWz5BgM2xsON1rVeqM">

You might be using an A1mini machine. This is because the print area of the A1mini is relatively small and cannot accommodate a pattern that is 0.001 steps long. You can try setting the starting and ending values a little smaller.

frenchiveruti, 2025-10-16:
> You might be using an A1mini machine. This is because the print area of the A1mini is relatively small and cannot accommodate a pattern that is 0.001 steps long. You can try setting the starting and [...]

Ah, that makes sense!

I would like to suggest an improvement then, if the printer is the A1Mini for it to say "You're using an A1 mini and the steps are either too small or start and ending are too wide"

qusheng-bambu, 2025-10-16:
> i think this does not need to be a machine-specific message. no need to mention A1 mini.
> 
> how about:
> 
> `Calibration pattern is too big for the build plate. Please reduce the k-value range and/or step size.`

Yes, thank you very much for your suggestion. We will consider adding relevant tips