bambulab/BambuStudio #7851: Cannot login in Online models in any way

kinczelt, 2025-08-13:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro 24H2

### Additional system information

Processzor: 12th Gen Intel(R) Core(TM) i9-12900K (3.20 GHz)
Memory:	32,0 GB 
VGA: RTX 3080

### Printer

Bambulap P1S

### How to reproduce

Open Bambustudio
Wait till Online models appear
Asks for login
Try loggig in (write in E-mail + password)
Ask for security code
Give security code, button turns grey
Nothing happens

Aalternatives:
1.
Google login click on
nothing happens
2.
Browser login click
nothing happens

### Actual results

Nothing

### Expected results

Login :D

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/21761894/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

qusheng-bambu, 2025-08-14:
Thank you very much for your feedback! I didn't reproduce this issue locally. This should be a probabilistic bug. Can it be solved by restarting Studio? Could you please upload the Studio log? This will help us troubleshoot this issue. We will do our best to solve it.
The steps to obtain the logs are as follows：
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

kinczelt, 2025-08-14:
No its not solvable via restarting the application, or the PC itself. Logs should be already available in the main ticket in log.zip thats all the log files the program created.

archerymirko-alt, 2025-08-20:
I have the same problem.
I hope, someone can help!

MackBambu, 2025-08-21:
Please try deleting the network plugin at C:/Users/USERNAME/AppData/Roaming/BambuStudio/plugins, then re-download it and restart Bambu Studio.

archerymirko-alt, 2025-08-21:
It woooooorks. Thanks sooo a lot!
Many greets from North of Germany.
Mirko

JohnSmithCYM, 2025-09-08:
Thanks for your feedback, we have found it is a bug and it will be fixed in the next release.
Before that, you can enable “Show history on the home page” to avoid this problem.

[image]