bambulab/BambuStudio #7853: Bambu Studio crashes when switching tabs sometime

thepartlow, 2025-08-13:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Printer

X1C

### How to reproduce

Sometimes when I am on the Device Tab and I click back to the Prepare Tab.

### Actual results

It freezes, then crashes.

### Expected results

It should go to the Prepare Tab.

### Project file & Debug log uploads

[log_iotc.txt](https://github.com/user-attachments/files/21763331/log_iotc.txt)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

qusheng-bambu, 2025-08-14:
Thank you very much for your feedback! I didn't reproduce this issue locally. Could you provide Studio logs to help us quickly locate and analyze the cause of the problem

The steps to obtain the logs are as follows：
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

thepartlow, 2025-08-14:
bambu-studio.exe.30156.dmp
<https://drive.google.com/file/d/1062IXuQ5AdN05fVnFRG6hjWwSfuJZy09/view?usp=drive_web>
 debug_Wed_Aug_13_23_42_33_29188.log.0
<https://drive.google.com/file/d/1q-QI8iCJShomqVaelJfRGJkRELPFVN77/view?usp=drive_web>