bambulab/BambuStudio #7854: Custom Colors are not Saved

Kiriohn, 2025-08-13:
### Bambu Studio Version

2.2.0 Public Release

### Where is the application from?

Bambu Lab Official website

### OS version

OS: Windows 11

### Printer

Bambu Lab A1

### How to reproduce

1. Install the latest updates (fresh install or just from update)
2. Click on the filament color selection on the Project Filament area. Add any Custom Color and hit OK.
3. Open the filament color selection again.

### Actual results

Custom color is not saved and would always be wiped out. Even when trying to do the same on the device tab on the AMS for setting the custom color on each of the filament it would stick only as long as the 3mf file is open. As soon as you close the file even when saving the changes when asked, once you open any file or the same file the custom colors are not saved.

### Expected results

You should be able to see the custom color that you set and should be there even after closing the 3mf file.

### Project file & Debug log uploads

[580.zip](https://github.com/user-attachments/files/21763576/580.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

fahaaaxiki, 2025-08-14:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you 
https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1

fahaaaxiki, 2025-08-14:
Do you mean that the custom color in the following figure is not saved in the "Custom colors” list?
[image]

Kiriohn, 2025-08-14:
> Do you mean that the custom color in the following figure is not saved in the "Custom colors” list? <img alt="Image" width="1442" height="824" src="https://private-user-images.githubusercontent.com/156769852/477939599-fb1d30d2-866c-41ac-8473-f62c6d1502e4.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTUxNjM3MTIsIm5iZiI6MTc1NTE2MzQxMiwicGF0aCI6Ii8xNTY3Njk4NTIvNDc3OTM5NTk5LWZiMWQzMGQyLTg2NmMtNDFhYy04NDczLWY2MmM2ZDE1MDJlNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgxNFQwOTIzMzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lNGFmOGY2MjQyN2ZjYjRkMTkyNzkxNDA4YmVkMDIwOTFmODIwODUxYTRjMTQ0OTdkOTdkNGFkNTZhZTg3ODRjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.oD5zWQyDQm3HTzPDMjvcOtOF1AyU8aVrQWUvKJISfqc">

Hi, yes. this is what I meant. It does not save on the list.

Tartare-Bambu, 2025-08-14:
Hi there, thank you for your feedback.
If you are referring to the pop-up shown above, we’re afraid it is not possible to save your custom color selection to the “Custom colors” area on the left. This is due to a system-level limitation: the dialog uses the default Windows color picker, and unfortunately our code cannot modify the behavior of this built-in dialog.

Kiriohn, 2025-08-14:
> Hi there, thank you for your feedback. If you are referring to the pop-up shown above, we’re afraid it is not possible to save your custom color selection to the “Custom colors” area on the left. [...]

I see, I guess I won't be moving to the latest update for now, the previous version works just fine. It broke when I updated. I've rolled back my version to the previous stable one already.
Hope you find a fix for this on the next update or so.

Kiriohn, 2025-08-14:
> > I see, I guess I won't be moving to the latest update for now, the previous version works just fine. It broke when I updated. I've rolled back my version to the previous stable one already. Hope you find a fix for this on the next update or so.
> 
> Hi there, sorry for the confusion earlier. After comparing with previous versions, we have identified the cause of the issue and will fix it in the next release. Thank you for bringing this to our attention, and we sincerely apologize for any inconvenience this may have caused. We will update you once the fix is available.

Thank you so much for taking the time to look into this. I will wait for the next patch/update fixing this issue.

Kiriohn, 2025-08-28:
Hi @Tartare-Bambu and @qusheng-bambu yeah sorry I just got back from traveling. I can confirm the bug was fixed. I will be closing this now. Thank you so much for your help!