bambulab/BambuStudio #7855: Unable to view 3d Prints, slice or print on the Desktp Application

criostage, 2025-08-13:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab github releases

### OS version

CachyOS (Kernel 6.16.0-7)

### Additional system information

CPU: AMD Ryzen 7 7700 8-Core Processor
Memory: 32Gb
Display:  AMD Radeon RX 7900 XT

### Printer

Bambu Lab P1S

### How to reproduce

1. Update BambuStudio to version 2.2.0.85 on Linux (unaware if this is an issue on other platforms) using the built-in update or flatpak
2. Open any Project file, saved locally or online

### Actual results

I wrote a post on Bambu Lab subreddit (https://www.reddit.com/r/BambuLab/comments/1mpefiz/bambu_studio_stopped_working_for_me/) with all the details and how for now to roll back to version 2.1.1.52.. but when opening any project file, i'm not able to see the render, unable to slice plate and therefore print from my Desktop linux machine, on mobile works. Same STL file works on version 2.1 but when upgrading bambu Studio to 2.2, the same file wont work (and i printed the file

### Expected results

Being able to open files and print them on the newer version 2.2. Note: as part of my testing, downgraded my app to version 2.1, deleted all files related to Bambu Studio, reinstalled it, deleted the files ... you name it. So the files you get below are from a clean installation, I haven't downloaded any profiles or even modified any setting in the application. Everything is as vanila as it get's. 

### Project file & Debug log uploads

[Piranha.zip](https://github.com/user-attachments/files/21763805/Piranha.zip)

[log.zip](https://github.com/user-attachments/files/21763806/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

af7v, 2025-08-14:
I can confirm the issue

System:
  Kernel: 6.8.0-71-generic arch: x86_64 bits: 64 compiler: gcc v: 13.3.0 clocksource: tsc
  Desktop: Cinnamon v: 6.4.8 tk: GTK v: 3.24.41 wm: Muffin v: 6.4.1 vt: 7 dm: LightDM v: 1.30.0
    Distro: Linux Mint 22.1 Xia base: Ubuntu 24.04 noble
  Memory: total: 16 GiB note: est. available: 15.37 GiB used: 6.25 GiB (40.7%)

Reported application version 2.2.0.85
Application installed via Flatpak from Flathub

```
:~$ flatpak info com.bambulab.BambuStudio

Bambu Studio - Get perfect 3D prints!

          ID: com.bambulab.BambuStudio
         Ref: app/com.bambulab.BambuStudio/x86_64/stable
        Arch: x86_64
      Branch: stable
     Version: 2.1.1 Public Release
     License: AGPL-3.0-only
      Origin: flathub
  Collection: org.flathub.Stable
Installation: system
   Installed: 737.7 MB
     Runtime: org.gnome.Platform/x86_64/48
         Sdk: org.gnome.Sdk/x86_64/48

      Commit: 7e7c2df12792ce6494c331d01309c9ed516f516b50d5d64846c546e591ae3994
      Parent: 89ea13aa4a965cfd6a5538503f905b16edc24a416e94d88f68cef734e7504807
     Subject: Reduce the number of concurrent jobs (603d9d0abe1e)
        Date: 2025-08-12 22:36:21 +0000

```

af7v, 2025-08-14:
To downgrade to 2.1.1.52 via flatpak
```
:~$ sudo flatpak update --commit=89ea13aa4a965cfd6a5538503f905b16edc24a416e94d88f68cef734e7504807 com.bambulab.BambuStudio
```

flamingm0e, 2025-08-14:
I'm sure someone will be along shortly to tell you not to use the "unofficial workaround" and to use the AppImage, and it's absolutely a worse experience than using the "unofficial workaround"

flamingm0e, 2025-08-14:
https://github.com/bambulab/BambuStudio/issues/7843#issuecomment-3186298225

cdamiens, 2025-08-15:
Hi, same issue here:
Kernel: 6.15.9-201.fc42.x86_64
OS: Fedora release 42 (Adams)
BambuStudio: 2.2.0.85 (Flatpak)

flamingm0e, 2025-08-17:
Check the latest version of Flatpak now.  It's working for me.