bambulab/BambuStudio #7857: Build Plate and imported models are not visible

silentjrules, 2025-08-14:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

Linux Mint 22.1

### Printer

Bambu Lab X1C

### How to reproduce

Open the Bambu Studio

### Actual results

No build plate or models are shown.

### Expected results

Build plate and models should be visible.

### Project file & Debug log uploads

[logs.tar.gz](https://github.com/user-attachments/files/21764431/logs.tar.gz)

### Checklist of files to include

- [x] Log file
- [ ] Project file

dbrckovi, 2025-08-14:
Does anyone have a suggestion what to do until this is fixed? Judging by project time line, development stopped in June (vacations maybe?). I doubt this will be fixed in the next week or so.
Should we downgrade to earlier version and disable auto update until then?

I'm on Zorin OS 17.3 (ubuntu 22.04 LTS)

jhob101, 2025-08-14:
I installed the [2.1.1 AppImage](https://github.com/bambulab/BambuStudio/releases/tag/v02.01.01.52)

after uninstalling the 2.2 version first.

silentjrules, 2025-08-14:
> I installed the [2.1.1 AppImage](https://github.com/bambulab/BambuStudio/releases/tag/v02.01.01.52)
> 
> after uninstalling the 2.2 version first.

Ok, I'll give the appimage method a shot.

joshpayette, 2025-08-16:
I received an update just now. After installing it, the issue is resolved. Version still shows 2.2.0.85, but a patch was offered in Software Center.

rotor-rig, 2025-08-16:
I had this just tonight on Kubuntu 24.04 . As Josh says, when I looked in software updates there was an update to my Flatpak version of Bambu Studio. Once installed, the build plate re-appeared. Version of Bambu Studio still says 2.2.0.85.

guanyun-gudujian, 2025-08-25:
not an rendering issue