bambulab/BambuStudio #7861: No load on Ubuntu 24.04

TomReed123, 2025-08-14:
### Bambu Studio Version

2.2.0.0

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 24.04

### Additional system information

Surface Pro 6. App image 7292 loads fine, 7829 won't load. Tried download twice.

### Printer

p1p

### How to reproduce

Try to open app.

### Actual results

nothing

### Expected results

A functioning app

### Project file & Debug log uploads

[log_iotc.txt](https://github.com/user-attachments/files/21764926/log_iotc.txt)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jhob101, 2025-08-14:
Also experiencing this issue on Zorin OS (ubuntu 22.02).  

App loads but no plates/models shown in Prepare and Preview, just a blank screen

[image]

Dartoxian, 2025-08-14:
Confirming I'm also seeing this on ubuntu 24.04. Downgrading to Bambu 2.1.1 mitigates the problem for me. I've installed with flatpak and needed to run:

```
sudo flatpak update --commit=62ca07241b7e88515dca0bdde3bc72d05fd7ff541dbb3d76205761ce5a4b244c com.bambulab.BambuStudio
```

BrainStone, 2025-08-14:
Can confirm the issues. When launching through the console I saw issues related to squashfs. Which seems to indicate that the AppImage build is just broken.

leoherzog, 2025-08-16:
Also issue here on Fedora 42 + Flatpak

n9sla, 2025-08-16:
Same issue for me Ubuntu 24.04

[image]

ThatOhio, 2025-08-16:
Same issue on CachyOS (KDE Plasma DE 6.4.4) running via the flatpak.

Knochi, 2025-08-16:
Same issue on Steamdeck which doesn't have NVIDIA gfx 

Operating System: SteamOS 3.7.13 Build: 20250630.1
KDE Plasma Version: 6.2.5
KDE Frameworks Version: 6.9.0
Qt Version: 6.8.1
Kernel Version: 6.11.11-valve19-1-neptune-611-g88b36d49a5e3 (64-bit)
Graphics Platform: X11
Partition: A
Processors: 8 × AMD Custom APU 0405
Memory: 14,5 GiB of RAM
Graphics Processor: AMD Custom GPU 0405
Manufacturer: Valve
Product Name: Jupiter
System Version: 1

luk-ada, 2025-08-16:
Same issue on Linux Mint, downgrading to Bambu 2.1.1 works.

```
RELEASE=22.1
CODENAME=xia
EDITION="Cinnamon"
DESCRIPTION="Linux Mint 22.1 Xia"
DESKTOP=Gnome
TOOLKIT=GTK
NEW_FEATURES_URL=https://www.linuxmint.com/rel_xia_whatsnew.php
RELEASE_NOTES_URL=https://www.linuxmint.com/rel_xia.php
USER_GUIDE_URL=https://www.linuxmint.com/documentation.php
GRUB_TITLE=Linux Mint 22.1 Cinnamon
```

doeffgek, 2025-08-16:
Was in another thread. Will be following up here.

ThatOhio, 2025-08-17:
> Same issue on CachyOS (KDE Plasma DE 6.4.4) running via the flatpak.

After updating my system and packages today (including a new update to the flatpak for bambu studio), I seem to be working as expected again.

jbwiden, 2025-08-17:
I uninstalled and reinstalled through the Ubuntu software center and it is back to working for me.

guanyun-gudujian, 2025-08-25:
not an rendering issue

posita, 2025-08-25:
> Problem resolved on my system. Used chmod x on .appimage. Haden't needed to do this for previous updates. Alos downloaded a new .appimage file.

This is not an .appimage problem. It's a flatpak problem.

TomReed123, 2025-08-25:
> > Problem resolved on my system. Used chmod x on .appimage. Haden't needed to do this for previous updates. Alos downloaded a new .appimage file.
> 
> This is not an .appimage problem. It's a flatpak problem.

Well since I origionated this post and am running .appimage perhaps it's a problem with both.

prodestrian, 2025-08-30:
Experiencing the same issue with the AppImage too, on Ubuntu 24.04. It does load eventually, but it's painfully slow.

It takes anywhere from 15-30 seconds to load since upgrading from the previous version.

I see this warning after a few seconds:

[image]

It disappears after about 10 seconds, then Bambu Studio appears but with a black screen. Around 5 seconds later I'm able to actually interact with the application.

I've experienced this with release 7289 and 8017 (2.2.1.60 from two days ago).

If I roll back to 7606 (2.2.0beta2) it loads within 5 seconds.