bambulab/BambuStudio #7876: New version of Studio is extremely slow and does not display any projects in the Prepare or Preview screens

codeserf01, 2025-08-14:
### Bambu Studio Version

v 2.2.0.85

### Where is the application from?

Bambu Lab github releases

### OS version

Mint 22.1 (ie. Ubuntu-based) - fully updated

### Additional system information

Operating System: Linux Mint 22.1 Cinnamon
Cinnamon Version: 6.4.8
Kernel: 6.8.0.55-generic
Processor: 13th Gen Intel iCore i7-13700 x 16
Memory: 64GB
Graphics card: Nvidia Device 2689 (Asus Prime Graphics Card - RTX 4070Ti Super)
Display server: X11
Display: 1920x1080

### Printer

Bambu Lab A1

### How to reproduce

This issue occurred immediately upon app update from Mint's Software Manager today (14August2025). (This default software manager makes available and installs Ubuntu's version of the Studio package, which is a Flatpack.) 

To reproduce:

1) Launch Studio after install/update. Note: Updated Studio will also install a 'network test' module on first run).
  problem 1: The app is EXTREMELY slow to start and respond generally

2) Open any project.
 problem 2: The project will fail to display in the Prepare AND Preview options. The basic build plate also fails to display 

### Actual results

The screen (right side) under the Prepare and Preview options remains completely blank after loading a project. This is not fixable.

Further:
1) Problem: There is no instruction or method found yet to roll back this package version to the previous (working) working version.

2) Problem : While a sample project (that previously worked) is included below, the Linux version application log is MISSING. It is not where your instructions (below) say it should be found ('$HOME/.config/BambuStudio/log').

### Expected results

The project object(s) should display in an editable state, along with the underlying build plate. 

### Project file & Debug log uploads

[dew shield_v3.3mf.zip](https://github.com/user-attachments/files/21776312/dew.shield_v3.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

orbotik, 2025-08-15:
Same, likely something glitchy with this build. Similar: #7877

MiguelAngelLV, 2025-08-25:
Same problem with AUR package in ArchLinux/Manjaro