bambulab/BambuStudio #7878: plate not appearing in the prepare stage

Helloyesthisisdog0830, 2025-08-14:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Built from source by yourself or others

### OS version

Linux Mint 

### Printer

P1s

### How to reproduce

Open the prepare step or try to import anything and nothing appears in the field

### Actual results

yup

### Expected results

something to show up

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

TJKwispy, 2025-08-14:
I am having the same issue. 
ryzen 5950x
AMD 7900xt
Bambu 2.2.0.85
Source: flathub

[debug_Thu_Aug_14_14_24_54_2.log.0.zip](https://github.com/user-attachments/files/21778411/debug_Thu_Aug_14_14_24_54_2.log.0.zip)

GabeNedden, 2025-08-14:
I have the same issue
Linux Ubuntu
V2.1.1

Pieandsim, 2025-08-14:
Same issue : 
- Fedora KDE
- Flathub

cmeans, 2025-08-15:
Same issue:
Fedora Gnome
Flathub

Reverted and stopped updates for now.

hatesigningup, 2025-08-15:
Same here.

Operating System: Nobara Linux 42
KDE Plasma Version: 6.4.2
KDE Frameworks Version: 6.15.0
Qt Version: 6.9.1
Kernel Version: 6.15.6-201.nobara.fc42.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 8 × Intel® Core™ i7-7700HQ CPU @ 2.80GHz
Memory: 16 GiB of RAM (15,5 GiB usable)
Graphics Processor 1: Intel® HD Graphics 630
Graphics Processor 2: NVIDIA GeForce GTX 1070 with Max-Q Design
Manufacturer: Micro-Star International Co., Ltd.
Product Name: GS73VR 7RG
System Version: REV:1.0

posita, 2025-08-15:
Probably a dup of #7861. Work-around (flatpak downgrade procedure) here: https://github.com/bambulab/BambuStudio/issues/7871#issuecomment-3190413022 .

Eikkargh, 2025-08-15:
Same issue on EndeavourOS. The workaround Flatpak downgrade process to 2.1.1 works.

MartinHerren, 2025-08-15:
Probably dupped by #7880, #7892, #7894, #7896, #7897, #7898 and probably more.

Helloyesthisisdog0830, 2025-08-15:
sudo flatpak update --commit=62ca07241b7e88515dca0bdde3bc72d05fd7ff541dbb3d76205761ce5a4b244c com.bambulab.BambuStudio

this worked for me to solve the problem even though it is a temporary solution

woody55uk, 2025-08-19:
> If the OP used Flatpak, then the latest build fixes it: https://flathub.org/apps/com.bambulab.BambuStudio and the bug can be closed.

I can confirm this. I had the issue and the latest update cured it for me. Thanks to anyone involved in the fix.