bambulab/BambuStudio #7882: P1进行动态流量校准时，值步长无法设置为0.001

iamapig120, 2025-08-15:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 24H2

### Printer

Bambulab P1S

### How to reproduce

起始K值 0.020
结束K值 0.050
值步长设置为 0.001 时报错，要求大于等于 0.001

### Actual results

被拒绝打印校准模型，要求大于等于 0.001

### Expected results

正常应该通过数值校对并允许打印

### Project file & Debug log uploads

No file here

### Checklist of files to include

- [ ] Log file
- [ ] Project file

iamapig120, 2025-08-28:
same as #7850

qusheng-bambu, 2025-08-28:
We have fixed this issue in the latest version. You can verify it with the latest version. If the verification is successful, you can close this issue

https://github.com/bambulab/BambuStudio/releases/tag/v02.02.01.60

closed 2025-12-02 by rouyuanzi-bambu