bambulab/BambuStudio #7885: Projects fails to load in Linux version 2.2.0.85 but will load in Windows version 2.2.0

Seldomscene01, 2025-08-15:
### Bambu Studio Version

Linux 2.2.0.85

### Where is the application from?

Bambu Lab github releases

### OS version

inux Mint 22.1

### Additional system information

AMD Ryzen 5 4600G with Radeon Graphics
32Gb Ram
AMD Radeon RX 580 2048SP (radeonsi, polaris10, ACO, DRM 3.57, 6.8.0-71-generic)
Kernel	Linux 6.8.0-71-generic (x86_64)
Version	#71-Ubuntu SMP PREEMPT_DYNAMIC Tue Jul 22 16:52:38 UTC 2025
C Library	GNU C Library / (Ubuntu GLIBC 2.39-0ubuntu8.5) 2.39
Distribution	Linux Mint 22.1

### Printer

A1 and A1 Mini

### How to reproduce

1. Go to file location
2.  Open File "Vosper MTB 234.3mf"
3. Dialog screen indicates it's loading file but nothing is rendered
4. Objects lists various stl files but nothing is rendered on the layout
5. clicking SLICE PLATE seems to slice plate object(s) but nothing apperas on the Prepare or Preview pages
6. 

[image]

[image]

[Vosper MTB 234.3mf.zip](https://github.com/user-attachments/files/21790503/Vosper.MTB.234.3mf.zip)

### Actual results

Objects seem loaded but nothing is rendered on the work space

### Expected results

It originally handled the project file - no Problem but as of today will not load any project file properly. Although, switching to windows on the same PC handles the files no problem. 

### Project file & Debug log uploads

[Vosper MTB 234.3mf.zip](https://github.com/user-attachments/files/21790506/Vosper.MTB.234.3mf.zip)

Only two files in the /home/dave/.config/BambuStudio/log folder:
debug_network_Sat_Dec_14_17_40_40.log.enc
&
debug_Sat_Dec_14_17_38_06_10036.log.0

and they are not accepted.

### Checklist of files to include

- [x] Log file
- [x] Project file

posita, 2025-08-15:
Probably a dup of #7861. Work-around as noted by @nocluewhatido (flatpak downgrade procedure) here: https://github.com/bambulab/BambuStudio/issues/7871#issuecomment-3190413022 .

Reini78wood, 2025-08-15:
I also do this dowegrade : sudo flatpak update --commit=ffd0eb5704c5e1e0af5592920551e5ce37f0d1c4a3a25928e249c11c4c5c5c7a com.bambulab.BambuStudio

it works

JohnPaulJones1747, 2025-08-15:
I have the same issue in Linux Mint, "Projects fails to load in Linux version 2.2.0.85".   However, I don't have a Windows machine to test that option.    

[image]

Seldomscene01, 2025-08-16:
Solved my problem by using the AppImage V2.1.1.52. Would like to know if
the V2.2.0.85 gets a fix so for now I'm happy to be back printing with
Linux as Windows wants to update to 11 and gives cranky messages about it.
Must work out if I can block its access to the internet (but not my
network) when I'm not using it for Bambu Studio. Thanks for the suggestions
folks.