bambulab/BambuStudio #7893: text tool

faithcarcrew, 2025-08-15:
### Bambu Studio Version

Bambu_Studio_win-v02.02.00.85

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab A1

### How to reproduce

open the text tool and try to center it, or just change the size as it gets buggy, you can no longer work the same way as before, you can't change the thickness to 0.01 it bugs every time you edit

### Actual results

open the text tool and try to center it, or just change the size as it gets buggy, you can no longer work the same way as before, you can't change the thickness to 0.01 it bugs every time you edit

### Expected results

that allows you to center the text, it is more practical, models that give this freedom to customize are becoming difficult because of this horror that the tool is

### Project file & Debug log uploads

[44+Magnum+50rnd+Complete_stls.zip](https://github.com/user-attachments/files/21794046/44%2BMagnum%2B50rnd%2BComplete_stls.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

GuyH77, 2025-08-16:
Just upgraded to the latest version of BS 2.2 and seen the changes to the text tool. Another frustrating step backwards. The ability to centre text on an object is now broken again. Please text, use the position object option and 0,0 now has the text off centre again. Please fix this.

faithcarcrew, 2025-08-17:
that's exactly it