bambulab/BambuStudio #7894: no 3d wision

VictorMolokanov, 2025-08-15:
### Bambu Studio Version

2.2

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Cachy os nvidia

### Additional system information

Nvidia GTX 1060 6GB

### Printer

It's not printer BUG!

### How to reproduce

1. update to 2.2 version
2. open
3. prepare
4. u see nothing

### Actual results

nothing in 3d view

### Expected results

plate in 3d view

### Project file & Debug log uploads

%

### Checklist of files to include

- [ ] Log file
- [ ] Project file

gartisk, 2025-08-15:
I've the same issue.
The bug happens for all projects
POP-OS 22.04 
NVIDIA 4070 Laptop GPU

[image]

[image]

posita, 2025-08-16:
Duplicate of #7861. Please close and subscribe to that issue.