bambulab/BambuStudio #7902: Printer selection suddenly disappeared

ivimail, 2025-08-15:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.6

### Printer

X1C

### How to reproduce

Suddenly, the option to select the printer on the slicer has disappeared. I can access the printer selector by pressing the wheel. I see that my model is selected, but I can't select it directly or change the nozzle or filament settings. Nothing.

[image]

### Actual results

Suddenly, the option to select the printer on the slicer has disappeared. I can access the printer selector by pressing the wheel. I see that my model is selected, but I can't select it directly or change the nozzle or filament settings. Nothing.

[image]

### Expected results

Suddenly, the option to select the printer on the slicer has disappeared. I can access the printer selector by pressing the wheel. I see that my model is selected, but I can't select it directly or change the nozzle or filament settings. Nothing.

[image]

### Project file & Debug log uploads

[debug_Fri_Aug_15_20_35_58_74801.log.0.zip](https://github.com/user-attachments/files/21800891/debug_Fri_Aug_15_20_35_58_74801.log.0.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

posita, 2025-08-16:
Duplicate of #7861. Please close and subscribe to that issue.

ivimail, 2025-08-16:
> Possible duplicate of [#7861](https://github.com/bambulab/BambuStudio/issues/7861)?

It's not the same problem. In my case, bambu studio works except for the problem that I can't choose the printer from the dropdown menu and their settings, as nozzle diameter. Also it's imposible to change settings because the left bar can't be resized, so you can't enable or disable any of the setings, but if you open or drop a model, you can see it, and laminate it.

If you delete the settings under /Users/xxx/Library/Application Support and reopen the program, everything works fine after reconfigure again bambu studio, like a new setup.