bambulab/BambuStudio #7906: 22.0.85 flatpak No Model Image!

tr0043t, 2025-08-16:
### Bambu Studio Version

22.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

Ubuntu

### Printer

A1

### How to reproduce

Latest flatpak version 22.0.85 displays NO model image at all. Nothing. Not even the vector markers. No image when importing new model, importing saved 3mf, creating new model, etc. Totally blank. Previous flatpak was working great, which was just a few days ago I believe.

### Actual results

No model image.

### Expected results

Model image.

### Project file & Debug log uploads

Not immediately available.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

monkeyslap06, 2025-08-16:
Same problem Linux Mint 22.1 with flatpak version 2.2.0.85. :(

chris-neely, 2025-08-16:
Seeing the same issue on Fedora 42

closed 2025-08-19 by fahaaaxiki