bambulab/BambuStudio #7921: New Build Plate Offset Feature cannot be selectively disabled

psiberfunk, 2025-08-17:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS

### Printer

H2D

### How to reproduce

Start a print with the newest H2D firmware (01.02.00.00)  / BambuStudio (2.2.0.85 as of this writing) and the new build-plate offset feature may trigger, even if the build plate is well aligned, as shown here:

[image]

[image]

### Actual results

There appears to be no way to turn this new "AI" feature off selectively without totally disabling build plate detection (which is useful and actually works to prevent issues, based on 2D barcode).  So i'm stuck with this glitchy feature that often gives false positives when the plate is fine, or completely disabling build plate detection (which I don't want to do).

### Expected results

I want to be able to turn this specific "offset" feature off without having to turn off build plate detection.  This new slicer/firmware iteration made things worse because now I have to choose between losing a useful feature (the 2D barcode detection) or getting false positives for the offset-positioning thing.. which seems to work poorly.  This kneecaps the "buildplate detection" feature.. there needs to be a separate checkbox disabling this offset detection thing until/unless it can be made to be way more reliable.

### Project file & Debug log uploads

NA

### Checklist of files to include

- [x] Log file
- [x] Project file

cseelye, 2025-08-21:
Even with the stock high temp/smooth PEI it triggers "detected build plate offset" almost every time. I don't want to turn off build plate detection because every once in a while I forget to replace the build plate or change it to the one I wanted, but if I leave it on then I get this error and have to force resume almost every print.

IanVS, 2025-09-03:
I get this offset error every time I try to print now.  Please either revert the feature if it's so buggy, or give us a way to turn it off.