bambulab/BambuStudio #7936: Calibration Tab not available

Janockel, 2025-08-19:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab A1, P1S

### How to reproduce

Open Bambu Studio...

### Actual results

In the Public Beta 2 (2.2.0.68) the Calibration Tab was available and PA Profiles could be added.
In this version there is missing the Calibration Tab.
Restart of the software and the printer and reinstallation did not solve the problem.

[image]

### Expected results

The Calibration Tab should be available.

### Project file & Debug log uploads

-/-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

simonth3dactions, 2025-08-19:
I have the same problem.

This happened to me after the newest update (version 2.2.0.85). I use and change PA profiles multiple times a week in my print farm, and now it has just disappeared.

I had it come back after a recent reset of my pc, do to just wanting to clean up. But now it is gone again, after just a couple of days.

[image]

fahaaaxiki, 2025-08-20:
Hello, we have identified this issue and are working hard to fix it. Please wait for the subsequent version update. We're very sorry for the inconvenience caused to you during use.

fahaaaxiki, 2025-08-20:
same as #7697