bambulab/BambuStudio #7939: Sidebar too small

DanielDieltiens, 2025-08-19:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 13.7.7

### Additional system information

3,4 GHz Quad-Core Intel Core i5
Radeon Pro 560 4 GB
8 GB 2400 MHz DDR4

### Printer

Bamu Lab A1

### How to reproduce

Open tab Prepare

### Actual results

Sidebar too small

[image]

### Expected results

readable sidebar

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/21858551/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

felsgaertner, 2025-08-20:
Same here.

Workaround: CTRL-W, in menu "View > Reset Window Layout"

[image]

illien3, 2025-08-20:
> hello,please t press "ctrl+w" key first if you want to change sidebar width,thanks If it still doesn't work, please find "Open Configuration Folder" in the Help menu, delete the file in the picture below, and then start the software again <img alt="Image" width="784" height="29" src="https://private-user-images.githubusercontent.com/132528822/478032703-12dcb84d-9433-409f-b44a-aa91bd621743.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU3MjczNjEsIm5iZiI6MTc1NTcyNzA2MSwicGF0aCI6Ii8xMzI1Mjg4MjIvNDc4MDMyNzAzLTEyZGNiODRkLTk0MzMtNDA5Zi1iNDRhLWFhOTFiZDYyMTc0My5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgyMFQyMTU3NDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNDEzZmYxN2I5M2FhZjQwMWZhNDcyMmQwMzRhYjM0YTZjZDlkNWY5YmM3ZjZhNjM0NTQwZTE2NGUxMjE5NjQ5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.eMUKqJIBSlmg_j_D-m4JIO8dHK-xVb5wrp-2tZ_vVik">
> 
> Hello, it is expected that the next version will automatically fix it. Currently, you can solve it through the above methods.

What side effects will deleting this .conf cause, and what will it remove? Really need clarifications before even considering updating to 2.2 as many of us had tons of user presets saved.

Haidiye00, 2025-08-21:
Hello, there are separate files preset by users, and generally configuration files only store global information such as user preferences.

[image]
Alternatively, you can also make this change
Make sure to operate the software in a closed state, thank you

Haidiye00, 2025-08-21:
Hello, we are very sorry, but we will continue to do optimization in the future to avoid such issues. Expected to be included in the next version

pfeerick, 2025-08-26:
> Hello, we are very sorry, but we will continue to do optimization in the future to avoid such issues. Expected to be included in the next version

@Haidiye00 A simple fix might be to "simply" expose the `enable_sidebar_resizable" parameter as an option in the "View" menu... i.e. "View -> Enable Sidebar Resize" which has a checkbox like "View -> Show 3D Navigator" when enabled. No need to manually edit (or delete!!) the file then ;)

Haidiye00, 2025-08-27:
Hello, please upgrade to the latest version
https://github.com/bambulab/BambuStudio/releases 

If there are still problems, we can solve them

closed 2025-08-27 by Haidiye00

DanielDieltiens, 2025-08-29:
Dear,

thanks I installed version 2.2.1.60

Before the A1 printer starts to print after having send a print file through Bambu Studio, I have to reselect the Print plate button before the printer starts processing the data.

Daniel