bambulab/BambuStudio #7942: AMS humidity shown in letters (A B ...)

felsgaertner, 2025-08-19:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab github releases

### OS version

Debian 12 (LMDE 6)

### Additional system information

My software source is Flathub, that is not listed as seletable source:

~~~shell
~$ flatpak info com.bambulab.BambuStudio

Bambu Lab - Get perfect 3D prints!

         Kennung: com.bambulab.BambuStudio
             Ref: app/com.bambulab.BambuStudio/x86_64/stable
     Architektur: x86_64
           Zweig: stable
         Version: 2.2.0 Public Release
         License: AGPL-3.0-only
        Ursprung: flathub
        Sammlung: org.flathub.Stable
    Installation: system
     Installiert: 731,8 MB
Laufzeitumgebung: org.gnome.Platform/x86_64/48
             Sdk: org.gnome.Sdk/x86_64/48

          Commit: eb766c24f5260c70cea99bf7f020834c14041f6c0fd00746a8a28fad4a557d8e
          Parent: cbd7f7736c12c833e4b4d219daa9008535a6e095b653b7f4336979196900587e
         Subject: Remove OSMesa dependency (7891c25ae0c1)
            Date: 2025-08-16 12:06:03 +0000
~~~

Firmware versions:
* Printer: 01.10.00.00
* AMS: 00.00.06.75

### Printer

X1C with AMS

### How to reproduce

1. Open Bambu Studio device page.
1. Select your printer with AMS

### Actual results

In Studio version 2.2.0.85 the AMS humidity level in the water drop is shown as A, B and so on.

### Expected results

In 2.1.1.52 the value was shown in percent, which I definitely prefer and expect.

Please restore the display as percentage value in Bambu Studio.

### Project file & Debug log uploads

Not required in this case.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

felsgaertner, 2025-08-25:
@fahaaaxiki: this is not a feature, it *is* a bug: 2.1.1.52 did show the humidity as numbers.

[image]

fahaaaxiki, 2025-08-26:
@felsgaertner Good idea, We will considerthis opinion If it can be achieved

Oliver-S3, 2025-09-13:
@fahaaaxiki  You know if you said more than "due to some reasons, we changed the display mode of the first - generation AMS back to the display by level" the forums wouldn't go nuts with conspiracy theories about an ulterior motive making the normal AMS less desirable than the AMS Pro.  It's fine to say the hardware isn't accurate enough or that you had to revert to an old revision due to unforeseen bugs.  Or maybe it is that you wanted to lower the feature set of the older model. 

Once a new feature is seen it can't be unseen all this does is fuel untrust.

felsgaertner, 2025-09-14:
@Oliver-S3 @fahaaaxiki Fun-fact on this discussion: the printer itself shows the humidity in numbers. I think it is not the hardware, just only related to the source code of Bambu Studio and therefore the question whether Bambu is willing to show numbers.

And Bambu, if you still insist in not showing measured values you should document the percentage ranges somewhere in your online manuals.

[image]

JohnSmithCYM, 2025-09-16:
Thanks for you feedback. I haved checked the reason with firmware group.
In AMS 1st generation, the humidity sensor is installed on the mainboard at the bottom of the AMS.
Because of its position, the sensor is affected by the heat from the printer chamber.
This leads to a significant deviation between the displayed relative humidity value and the actual humidity inside the AMS.

For AMS 2nd generation, the humidity sensor is placed inside the AMS chamber itself, so it is not influenced by the printer’s chamber temperature and provides accurate relative humidity readings.

（That said, we recognize that showing the value in percentage form may still be valuable for some users (even though it is not reliable in the scenario described above). We are exploring ways to present it as a converted percentage value. If there are any updates on this matter, we will share them here.）

And it changed on H2D firmware by now, and X1C firmware will update it in the future.