bambulab/BambuStudio #7945: macOS: H2D Virtual Camera fails to start due to missing libBambuSource.dylib linkage

DadeMurphy72, 2025-08-20:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 26 - but on previous builds as well

### Printer

Bambu H2D

### How to reproduce

Go Live for video feed then attempt to open ffmpeg in a stream

### Actual results

Live stream does not open

### Expected results

Live stream should open

### Project file & Debug log uploads

Not needed, I figured out what was happening.

Need to change bambu_source to link against @loader_path/libBambuSource.dylib instead of @rpath/libBambuSource.dylib. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DadeMurphy72, 2025-11-12:
I deleted my OBS setup and camera tools. I installed this build, installed camera tools, set up OBS and the live stream appeared as it should. Looks like that did the trick, great job all!

Bambu-Pomfret, 2025-12-04:
Hi @DadeMurphy72,

Great to hear that it resolved the issue.

Just a quick update: we've officially fixed this issue in our latest official release. To ensure you get the most stable experience, could you please give the newest version a try when you have a chance?

Thanks for your patience and feedback - they are crucial for us to keep improving!