bambulab/BambuStudio #7951: Side Navigation Bar Size

petrichorponds, 2025-08-20:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

Sequoia 15.6

### Printer

A1 Mini

### How to reproduce

1. Open Bambu Studio
2. Side Nav Menu is "minimized"; need to hit command+W to "reopen"

[image]

### Actual results

Restores side menu to "full size" but only until application is closed.

### Expected results

I want it to stay "open" permanently and not have to reset it every time I open this application.

[image]

### Project file & Debug log uploads

....

### Checklist of files to include

- [ ] Log file
- [ ] Project file

felsgaertner, 2025-08-20:
Duplicate of #7939, for workaround refer to the comments.

Haidiye00, 2025-08-21:
hello,please t press "ctrl+w" key first if you want to change sidebar width,thanks
If it still doesn't work, please find "Open Configuration Folder" in the Help menu, delete the file in the picture below, and then start the software again
[image]

closed 2025-08-21 by Haidiye00