bambulab/BambuStudio #7961: Left Side Bar is Squished Permanently Hiding Key Abilities

churdman001, 2025-08-21:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.6

### Printer

P1S

### How to reproduce

1. Open Bambu Studio
2. Go to the Prepare tab
3. Note that the left hand option pane is very small, hiding key features like the ability to change printer type

### Actual results

[image]

### Expected results

For the side bar to be normal size

### Project file & Debug log uploads

[Big.3mf.zip](https://github.com/user-attachments/files/21920952/Big.3mf.zip)

The zipped log file is 39.8MB and this copy past feature won't let me load it.

### Checklist of files to include

- [ ] Log file
- [x] Project file

EricDimitri, 2025-08-21:
@churdman001 , have you tried making it wider?

Haidiye00, 2025-08-22:
hello,please t press "ctrl+w" key first if you want to change sidebar width,thanks
If it still doesn't work, please find "Open Configuration Folder" in the Help menu, delete the file in the picture below, and then start the software again
[image]

Please make sure to use the latest version of the software and operate the Bambustudio. conf file while the software is closed
We will do another optimization in the next version, sorry for the inconvenience.

closed 2025-08-22 by Haidiye00