bambulab/BambuStudio #7980: Reload from Disk not working if .3mf file is closed and reopened

project-x51, 2025-08-24:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

I have raised this issue before and it was fixed. It is broken again.

### Printer

X1 Carbon

### How to reproduce

1. Add a step model
2. Save the project as a .3mf
3. Close Bambu Studio
4. Update the step model (I generate it from Solidworks)
5. Open Bambu Studio
6. Attempt to reload the model from disk
7. Error "Unable to reload:"

[image]

### Actual results

Cannot reload the model. Can replace with STL.

It would also be nice if the Reload From Disk menu option was available if I select more than one model.

### Expected results

It should update the model with the changes. Will work if I do this without closing Bambu Studio first.

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-08-25:
@project-x51 Thank you for your feedback. Could you please provide the 3mf file?

Haidiye00, 2025-09-08:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)
and can you provide your 3mf?thanks