bambulab/BambuStudio #7984: (CORE FW ISSUE) (H2D+X1C) part cooling fan spins down before directed by gcode

malventano, 2025-08-25:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

H2D and X1C

### How to reproduce

Temperature tower overhangs were noted to be significantly worse than prior experience. Watching the print, it appeared the part cooling fan was **spinning down** just prior to the left overhang. This made no sense as it was the opposite of what it should do. The part fan was going to full speed, but just not at the correct times, and not as often as it should. I confirmed gcode was correct and created a sample project to diagnose further, which led to uncovering the issue was far worse than expected, and this explains much of the overhang weirdness (failures, curling, etc) that I've seen lately.

I have created an extreme example (project attached) that shows the issue is far worse than I had originally thought (fan spins down before it even spins up for the overhang):

[image]

### Actual results

The above is a simplified version of the temperature tower shape and left side overhang. Pre fan start time of 1 second, combined with slow outer wall speed, means the fan has over a second to spin up, but watching the printer in action, the part cooling fan barely changes speed, actually slows back down before reaching the overhang, and prints the entire overhang at 10% (where it should be 100%). The picture above shows Bambu Studio is making the correct gcode (M106S255 (100%) is issued at the rear right corner, and M106S25.5 (10%) is not issued until *after* the overhang).

Since I noted [some new bugs impact firmware on both H2D and X1C](https://github.com/bambulab/BambuStudio/issues/7693#issuecomment-3184241084), I then replicated the above on the X1C with the same results. In the case of the X1C, the part cooling fan barely hits 20% while printing the rear wall, and prints the overhang at 10% fan speed. On prior firmware, this X1C part cooling fan always spun up in ~1 second and did not spin down until the end of an overhang, as expected.

### Expected results

Part cooling fan should follow gcode directives. It appears the newest firmware (on both H2D and X1C) are spinning down the fan sooner than the gcode has directed, to the point where for small overhangs, the overhang fan speed becomes irrelevant, even with fan pre-fan start time in use. This leads to poor overhangs, curling, etc.

### Project file & Debug log uploads

[H2D and X1C project files for part cooling fan issue.zip](https://github.com/user-attachments/files/21959642/H2D.and.X1C.project.files.for.part.cooling.fan.issue.zip)

Exported logs from both the H2D and X1C have been archived and are available upon request.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

malventano, 2025-08-25:
I did another test print on both H2D and X1C with a 2x long path before the overhang. Part cooling fan had enough runway to ramp up to 100%, but it then started dropping back to 10% **before the start of the overhang**, and appeared to be back to 10% by the end of a 10mm long overhang.

hotellonely, 2025-08-25:
> I did another test print on both H2D and X1C with a 2x long path before the overhang. Part cooling fan had enough runway to ramp up to 100%, but it then started dropping back to 10% **before the start of the overhang**, and appeared to be back to 10% by the end of a 10mm long overhang.

Maybe do a couple videos if you can?

malventano, 2025-08-25:
> Maybe do a couple videos if you can? 

Original (smaller) part where the fan does not spin up at all:

https://youtu.be/DFYM8ENSo1U?si=bQ5APmAkvogEMr-t

Longer part with more time to spin up the part cooling fan, but it slows back down before reaching the overhang:

https://youtu.be/yRFauiji-PA?si=0gJml698xIAHSsQf

hotellonely, 2025-08-25:
> Original (smaller) part where the fan does not spin up at all:
>
> https://youtu.be/DFYM8ENSo1U?si=bQ5APmAkvogEMr-t
>
> Longer part with more time to spin up the part cooling fan, but it slows back down [...]

You should also submit a ticket from Bambu's customer support, and upload all the proof/materials overthere as well.

malventano, 2025-08-28:
> [@malventano](https://github.com/malventano) Thank you for your feedback. This is a firmware issue. Please follow the instructions below to create a post-sales order. The firmware team will resolve the problem based on the post-sales order. The firmware team does not track orders on GitHub. Sorry.

Recommend opening a github section/tag for the reporting of firmware issues. There are currently many issues such as this where a simple description and/or video is more than adequate to demonstrate the issue. Having extra impediments to the process is going to slow Bambu's ability to correct them in a timely manner. Users will be hesitant to take their printer down just to repeat the issue and submit logs, etc. This means some valid issues will go unreported.

hotellonely, 2025-08-29:
> Recommend opening a github section/tag for the reporting of firmware issues. There are currently many issues such as this where a simple description and/or video is more than adequate to demonstrate [...]

you're talking to people taking saleries not people taking stock shares... ;)

malventano, 2025-09-21:
Additional information from support:

```
The feedback I received from the R&D department is that this phenomenon is related to the underlying design of the motion controller, and it could be very challenging for us to change, and we might not have a plan to change it currently.
```

Given this information, it would be on Bambu Studio to implement a workaround. Similar to the part fan spin up x seconds sooner, we would need the ability to spin down the fan (any value lower than the current value) x seconds *after* the expected location. This would ensure the part fan remains at overhang/bridge speed until the overhang/bridge is completed.

benjdero, 2026-09-22:
@BoBoYu-BamBu @ChloeShang123 Does that means your 2000$ top end printer can't do overhangs properly?