bambulab/BambuStudio #7989: Manual Flow Dynamics Broken

brandonraineri, 2025-08-25:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

H2D

### How to reproduce

Run manual flow dynamics and enter values and hit calibrate

### Actual results

[image]

### Expected results

Calibration should start without error

### Project file & Debug log uploads

No project file needed for manual calibration

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-08-26:
@brandonraineri  Thank you for your feedback. This issue has been resolved and verified in the internal version. Please look forward to the next version.

BoBoYu-BamBu, 2025-08-27:
@brandonraineri  Hello, please update to our latest version V2.2.1.58 to verify this issue. The download link is: https://bambulab.cn/zh-cn/download/studio